RetroPie/RetroPie-Setup #1337: ScummVM Mouse Issue

Ltty, 2016-03-05:
Not sure if that's the right place to post this bug, but in the latest release of retro pie, the mouse automatically moves into the left upper corner and therefore is unable to use. This bug seems to be related to a similar issue on Windows where a certain SDL version is causing this issue.

Does anybody know how to solve it?

joolswills, 2016-03-05:
in the scummvm gui or ? Our scummvm is still built on SDL 1.2.x and nothing has changed with the binaries in the last version. I just loaded scummvm and the mouse pointer is moving fine in the gui.

Ltty, 2016-03-05:
Yes, in the ScummVM GUI as well as in the games. I can move the mouse but
it automatically moves into the upper left corner.

Is it possible that this is hardware related, I'm using a brand new Pi 3.
Also tried with a clean install and different input devices.

joolswills, 2016-03-05:
Don't have that trouble. Maybe you have a joystick plugged in that is causing the mouse to move ?

joolswills, 2016-03-05:
(or the mouse is not working correctly with the rpi3)  - but I cannot reproduce.

Ltty, 2016-03-06:
A keyboard was the only other controlling device connected to the Pi.
Removed the keyboard, tried various wireless and wired mouses without
success. The mouse pointer automatically moves into the upper left corner.

I can use mouse and keyboard in other emulators though, Quake 3 Arena works
for example.

I found a ScummVM bug report with the same issue on Windows here:
http://forums.libsdl.org/viewtopic.php?t=6023

I think it's related to the Pi3, don't have the issue on a Pi2 with the
same hardware.

joolswills, 2016-03-06:
I don't think that issue is related. Do you have a sufficient PSU on the rpi3 ? Have you tried another PSU ?

joolswills, 2016-03-06:
Unfortunately it is working fine here, so I am not sure what I can do further.

HerbFargus, 2016-03-06:
Just tried the latest build and no issues. some must be something on your end.

joolswills, 2016-03-06:
the upcoming version (will have binaries uploaded shortly) will use SDL2 instead of SDL1 o you can try that and see if it helps, but if not I suspect it's a problem related to hardware or firmware incompatibility or something.

joolswills, 2016-03-08:
scummvm has been updated to v1.8 - you can see if you have the same problem. It uses a different version of SDL, but since we can't reproduce the issue and no-one else has reported I am closing.

closed 2016-03-08 by joolswills

Ltty, 2016-03-08:
Thanks for the support. The issue does not appear anymore after the update.

*--*

*Florian Lettner | *CEO & Co-Founder

*Pocket Lifestyle IT-Solutions GmbH*

Softwarepark 26 | 4232 Hagenberg, Austria

florian@pocket-lifestyle.com

www.pocket-lifestyle.com

*Von:* Jools Wills [mailto:notifications@github.com]
*Gesendet:* Dienstag, 8. März 2016 16:06
*An:* RetroPie/RetroPie-Setup <RetroPie-Setup@noreply.github.com>
*Cc:* Florian Lettner <florian@pocket-lifestyle.com>
*Betreff:* Re: [RetroPie-Setup] ScummVM Mouse Issue (#1337)

scummvm has been updated to v1.8 - you can see if you have the same
problem. It uses a different version of SDL, but since we can't reproduce
the issue and no-one else has reported I am closing.