RetroPie/RetroPie-Setup #1339: Emulators to be moved out of experimental menu

ProxyCell, 2016-03-05:
For some time now there have been some emulators that certainly appear to be very stable and have been widely reported to be working, even on Jessie

Off the top of my head I can think of at least lr-beetle-vb and lr-gw - but there are also some SNES/NES/GBA ones too

https://github.com/RetroPie/RetroPie-Setup/issues/1029 has a bunch of great test results at the bottom for things that compile and execute well - is there anything else that is required to be tested/developed for these to be moved out?

HerbFargus, 2016-03-05:
Beetle VB crashes with savestates and also has some errors in the saving paths.

joolswills, 2016-03-05:
nothing is really going to move out of experimental until we have our new modular system in place - plenty of things in experimental are stable, but we don't want to include them all in the default image. They will become "optional" packages, but will be installable from binary. Our current system doesn't really allow for this.

ProxyCell, 2016-03-07:
thank you both for that information, I couldn't tell from my end really since they all appeared to be really stable and just as ready to go.

closed 2016-03-07 by ProxyCell