RetroPie/RetroPie-Setup #1345: Native support bluetooth for Dualshock4 controllers.

Enari, 2016-03-06:
Since the Dualshock4 controller is a really popular controller native support is a must.
Now that we have **BlueZ 5.14+** pairing the controller works naively. Sadly due to a kernel bug introduced in kernel 3.15 most, if not all DS4 controllers do not work over bluetooth without the use of **ds4drv**.

This bug has been fixed as of kernel 4.5rc1. But lets not wait for that. 
We could supply a patched kernel module with the pre-built retropie images. 
And DS4 controllers would work naively for everyone.

Link to the issue reported for Ubuntu: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1511855

joolswills, 2016-03-06:
You need to speak to raspberrypi/linux devs if you want something backported - we don't want to get into maintaining kernels. RetroPie-Setup is designed for installing on top of an existing OS - our image is just that - a Raspbian Lite with RetroPie-Setup installed on it (And a few config tweaks).

joolswills, 2016-03-06:
https://github.com/raspberrypi/linux

Enari, 2016-03-06:
Yeah, I get it. It's not trivial compiling kernel modules...

closed 2016-03-06 by Enari