RetroPie/RetroPie-Setup #1349: No configuration generated in /opt/retropie/configs/all/retroarch-joypads after controller setup in ES

tdicola, 2016-03-07:
I'm having a really hard time getting the latest Retropie 3.6 image on a Raspberry Pi 3 to work with any controller and it appears there's some issue with it not generating configurations for retroarch.  I've tried an Xbox360 controller & official wireless dongle, a PS3 controller, and a generic USB gamepad (this one from Hyperkin: http://www.amazon.com/Hyperkin-USB-Pixel-Controller-Gray-Famicon/dp/B00G24ZHCY) and they all have the same problem.  I can configure them just fine in Emulation Station and control the UI.  However as soon as I launch a retroarch game I see the yellow text saying none of the controllers could be configured and sure enough the controller doesn't work at all in the game.

From digging around various bugs, forums posts, wiki pages etc. I'm pretty lost as to what could be wrong.  However according to this wiki page: https://github.com/retropie/retropie-setup/wiki/retroarch-configuration I should see configuration files in /opt/retropie/configs/all/retroarch-joypads.  I see nothing in that folder, which is probably why retroarch can't find any controllers.  Shouldn't I see some input files generated after ES configures a joypad?  Is there any way to manually trigger whatever process creates the files here?

joolswills, 2016-03-07:
did you manually alter any files at all after install ?

please post a copy of (via pastebin or similar)

/home/pi/.emulationstation/es_temporaryinput.cfg
/home/pi/.emulationstation/es_input.cfg

you can manually try and trigger generation by running from terminal:

```
/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh
```

nicholaus12, 2016-03-07:
I updated from 3.5 to 3.6 and noticed that I can no longer save core input for retroarch. For whatever reason, my player 1 was configured correctly, but when trying to change remaining players, it would save for the current instance of the rom, but when backing out and going back in, it would default back to what it was before. Not sure if this applies to OP's problem. I checked the systems retroarch configs and saw they weren't being saved.

joolswills, 2016-03-07:
@nicholaus12 this is unrelated - but please use the forum first for support as it may be a configuration issue.

tdicola, 2016-03-07:
Huh weird, I tried things from scratch again and it seems to be working now.  One thing I noticed that I didn't do before is explicitly restart emulation station after setting up the controller (i.e. using the menu option to restart).  I noticed it prints some text about configuring retroarch, etc.  Before I was just logging into the pi and issuing a reboot command while ES was running.  Perhaps I wasn't doing a proper restart and the configure script wasn't being called?  In any case I'll close this for now as I can't repro it.

closed 2016-03-07 by tdicola

joolswills, 2016-03-07:
the file are generated right after you configure the controller

hheimbuerger, 2016-12-22:
I had the same issue in 4.1. I suspect that in my case the issue was caused by the first controller that I tried to set up being the Steam Controller, but I hadn't yet figured out that the driver needs to be installed first (the changelog makes it sound like the support is built-in, to be honest).

So this apparently somehow got the system into a weird state, in which I had three different controllers set up (I could see them all in `/home/pi/.emulationstation/es_input.cfg` and they worked in EmulationStation), but the mappings weren't active in games, and every time I finished the controller setup, I immediately got back to the ES main screen. Now I know that the system should always freeze for a couple of seconds after setting up a controller, while it's rebuilding the RetroArch configuration, I guess.

Also, `/opt/retropie/configs/all/retroarch-joypads` was and had always been empty.

What fixed it for me was BuZz' hint in https://retropie.org.uk/forum/topic/5069/resetting-controller-configuration/2: I quit ES, ran `sudo ~/RetroPie-Setup/retropie_setup.sh`, went to `Manage Packages -> Core Packages -> emulationstation -> Configuration` and chose the option `Clear/Reset Emulation Station input configuration`. Restarted ES and from there on I got the freezing after setting up a new controller and all controllers started to work as configured in games, too.

alexhenrie, 2016-12-28:
I had the same problem with a Mayflash GameCube Controller Adapter working in the emulation station but giving a "controller not found" error whenever I started an emulator. `/opt/retropie/configs/all/retroarch-joypads` was empty, and I had tried deleting `.emulationstation/es_input.cfg`, but nothing fixed the problem until I reset the controller configuration according to the instructions in the previous post.

rchadwick82, 2017-01-31:
Thank you so much for this. I was pulling my hair out trying to get my controllers to work in the emulators after writing a new image to my sd card. I tried deleting all the config files, rebooting and setting up the controllers in emulation station multiple times with no luck. Then I found this thread, followed hheimbuerger's instructions and it's now working, thank you!!

bernhard263, 2020-07-12:
Resetting the configuration did not help me, but the post from joolswills (manually run the input generator script) gave a very helpful hint.

Here is what should normally happen: When you have finished setting up your controller in emulationstation, it will generate a file ``es_temporaryinput.cfg`` in ``~/.emulationstation``, then it will run this script:
````
/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh
````
The script will read the temporary input file and generate a retroarch joypad config file from it in ``/opt/retropie/configs/all/retroarch-joypads``

But for some reason, it never generated the temporary input file on my machine, so the script had no effect when it ran. I then tried to manually create a es_temporaryinput.cfg from my already existing es_input.cfg and ran the script again:
````
cp ~/.emulationstation/es_input.cfg ~/.emulationstation/es_temporaryinput.cfg
/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh
````

Then checked my retroarch joypad folder and sure enough, this time it did create the joypad file:
````
ls /opt/retropie/configs/all/retroarch-joypads
Xbox 360 Wireless Receiver.cfg
````

cmitu, 2020-07-12:
@bernhard263 please use the forum for support questions.

bernhard263, 2020-07-12:
> @bernhard263 please use the forum for support questions.
This wasn't a question. The OP suspected it might be a bug in one of the RetroPie scripts, and I provided my findings as to what part of it isn't working as expected because it might help solve the problem.

bernhard263, 2020-07-15:
It turns out that it wasn't a bug, it was my fault. After using the reset input option, I deleted the es_input file manually because the D-pad was misconfigured and I couldn't navigate the menu. I figured that wouldn't be a problem because emulationstation would recreate it anyway. But when the file is created from scratch by emulationstation instead of the RetroPie setup script, it is missing some custom code that will trigger the config generation for retroarch every time. From the manual:
````
RetroPie uses custom generation scripts that will generate controller configurations for many emulators through the initial ES controller configuration, so if you remove or make any edits to this configuration you need to regenerate that hook
````

mrworf, 2020-10-02:
I had the same issue and the tip by @hheimbuerger didn't help. However, when trying to edit the file manually, I found out that I simply didn't have any more space on the SDcard (actually a SSD via USB3).

It should have expanded to use all the space, but it turns out that the `raspi-config` option of expanding partitions doesn't know how to handle non-SDcards (such as my SSD). 

Expanding the partition using `gparted` on another machine which doesn't use the SSD as boot worked and now RetroPie is able to write configuration options and all is well.