RetroPie/RetroPie-Setup #1350: Enhancement for DosBox: SDL2, DynRec

jimbo27, 2016-03-07:
I found an enhancement for DosBox. There is a patch which provides SDL2 which should give some speed improvements when used SDL2.0.4.
Besides this enabling DynRec should give additional speed according to this thread:

https://www.raspberrypi.org/forums/viewtopic.php?f=78&t=137974

> The patch you need is here:
> http://www.vogons.org/viewtopic.php?t=34770
> 
> Go to the last pages for most recent version.
> You caan apply it like this on the dosbox sources dir:
> patch -p0 < ../dosbox_sdl2_20150713.diff
> 
> That will give you SDL2 support. NOTE you must have previously built and installed SDL 2.0.4 since it's not yet available on Raspbian repositories!
> Then you can enable dynarec support by setting these like this in config.h:
> #define C_DYNREC 1
> #define C_TARGETCPU ARMV4LE
> #define C_UNALIGNED_MEMORY 1
> 
> Then, set cpu type to dynamic in your .conf file, and you can easily aim for a 2000 cycles system without worrying about sound dropouts. :)

http://www.vogons.org/viewtopic.php?f=32&t=34770&start=56

Perhaps this is interesting enough to give it a try...

joolswills, 2016-03-07:
we already have DYNREC enabled.

I don't think there will be any speed improvements using sdl2 - but it would be better to move to it - I will have a look.

amigagamer, 2016-03-14:
This is an old thread but should be useful.  See post 4 regarding the scaling settings for sdl2
[https://www.raspberrypi.org/forums/viewtopic.php?f=78&t=110957&p=761752#p761752](url)

Looking at my working config file from sdl2 build (r3910 may 2015) i recall lots of time spent trying to get sdl2 to handle the scaling.  To get this working i used the following settings in dosbox.conf, although keep in mind this was a 12 month old svn/version of the patch.  Hopefully the information is useful!

[sdl]
fullscreen=true
vsync=false
fullresolution=0x0
windowresolution=original
output=texture

[render]
frameskip=0
aspect=false
scaler=none

psyke83, 2017-12-29:
There's now an experimental dosbox-sdl2 scriptmodule - I think we can close this issue.

closed 2017-12-29 by psyke83