RetroPie/RetroPie-Setup #1351: Gamelist-only and automatic restart not working after 3.6

IDmedia, 2016-03-08:
Downloaded the Stanard 3.6 image and noticed that editing "10-emulationstation.sh" and adding "--gamelist-only" doesn't work anymore. Now all files are listed regardless if the argument is set or not. Where should the argument be set now in order to take effect?

Another "issue" is that RetroPie doesn't seem to restart ES anymore. Previously, if I exited ES it would restart after a few seconds unless a key is pressed. Is it currently broken or has this been changed for some reason?

joolswills, 2016-03-08:
it is changed - there is an option in ES to quit/shutdown/restart now. The parameters should be sent through still to emulationstation but I'll check that. Have you made any modifications apart from that ?

IDmedia, 2016-03-08:
I haven't made any changes that would affect that. When I choose "Quit EmulationStation" is closes immediately and doesn't respawn.

joolswills, 2016-03-08:
that is correct - you should have a restart emulationstation option to restart it.

IDmedia, 2016-03-08:
Ok, so there is no easy way of turning the feature back on that ES restarts when you choose "Quit" by mistake? I liked it when it restarted automatically so I don't have to find a USB keyboard every time I manage to hit the wrong button... but I guess I'll make my own restart script or remove the menu entry. Thanks.

joolswills, 2016-03-08:
No - there were too many complaints - hence now you have control whether it quits / restarts. It's not that easy to quit by mistake with the interface since it asks you to confirm

joolswills, 2016-03-08:
you choose quit -> then from the next menu you have to choose again, and then you get a confirmation. I don't think we can be safer than that without it being silly.

closed 2016-03-08 by IDmedia