RetroPie/RetroPie-Setup #1355: lr-ppsspp - Segmentation fault with emulationstation but ok in command line.

salcin, 2016-03-08:
Hi,

I do not understand when i run a psp game from emulationstation I get the error in /tmp/runcommand.log

`/opt/retropie/supplementary/runcommand/runcommand.sh: line 791:  4638 Segmentation fault      (core dumped) /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-ppsspp/ppsspp_libretro.so --config /opt/retropie/configs/psp/retroarch.cfg "$ISO" --appendconfig /tmp/retroarch.cfg`

But if I launch from my terminal with the command below, the game works with somes warnings :

$ /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-ppsspp/ppsspp_libretro.so "$ISO"
W: file_util.cpp:215: W: IsDirectory: stat failed on /home/user/RetroPie/roms/psp/PPSSPP/lang/.ini
I: gl_state.cpp:137: I: GPU Vendor : NVIDIA Corporation ; renderer: GeForce Go 7300/PCIe/SSE2 version str: 2.1.2 NVIDIA 304.131 ; GLSL version str: 1.20 NVIDIA via Cg compiler
I: gpu_features.cpp:20: I: Checking for GL driver bugs... vendor=1 model='GeForce Go 7300/PCIe/SSE2'
W: file_util.cpp:215: W: IsDirectory: stat failed on /home/user/RetroPie/roms/psp/PPSSPP/flash0/font/jpn0.pgf
Failed to load ppge_atlas.zim.

Place it in the directory "assets" under your PPSSPP directory.
W: file_util.cpp:215: W: IsDirectory: stat failed on /home/user/RetroPie/roms/psp/PPSSPP/shaders
I: zip_read.cpp:370: I: Not a VFS path: /home/user/RetroPie/roms/psp/PPSSPP/PSP/shaders . Reading local directory.
W: file_util.cpp:215: W: IsDirectory: stat failed on /home/user/RetroPie/roms/psp/PPSSPP/flash0/font/jpn0.pgf
^CW: file_util.cpp:215: W: IsDirectory: stat failed on /home/user/RetroPie/roms/psp/PPSSPP/flash0/font/jpn0.pgf

How remove the parameters --config /opt/retropie/configs/psp/retroarch.cfg and --appendconfig /tmp/retroarch.cfg and why my PPSSPP directory is /home/user/RetroPie/roms/psp/PPSSPP and not /home/user/RetroPie/BIOS/PPSSPP ?

Thank you in advanced.

joolswills, 2016-03-08:
probably because you have some core options that have been changed and are not being used when not using retroarch.cfg - but they are causing problems. remote associated option in /opt/retropie/configs/all/retroarch-core-options.cfg

launch parameters are stored in /opt/retropie/configs/ppaapp/emulators.cfg

joolswills, 2016-03-08:
appendconfig is added automatically by runcommand. it only adjusts render res etc.

joolswills, 2016-03-08:
for other questions regarding PPSSPP please use the forum (this issue tracker is used for tracking retropie bugs - https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md)

salcin, 2016-03-08:
Thank you for your quick response.

Sorry for the inconvenience, I thought this was a bug and not a problem of configuration.

Thank you for your work.

closed 2016-03-08 by salcin