RetroPie/RetroPie-Setup #1356: Another line 11 "no such file or directory" problem

andyfrueh, 2016-03-08:
I just performed a binary refresh from a formerly working RetroPie install.  Now when I reboot, ES fails to load.  It's looking for the EmulationStation executable at /opt/retropie/supplementary/EmulationStation but, sure enough, there's no file named "EmulationStation" at that location.

Any ideas what may have went wrong, and how I can correct it?

HerbFargus, 2016-03-08:
Sounds like a network issue. Tryinstalling it again.

joolswills, 2016-03-08:
did you update the retropie-setup script first ?

andyfrueh, 2016-03-08:
I did update the setup script.  Trying again.  I noticed that, during the install of the SDL 2.0.1 binaries, I'm getting a 404 error.  Not sure if that's related at all, but sounds like maybe a dependency is missing?

joolswills, 2016-03-08:
sdl 2.0.1 -  what system / os are you on? - we currently ship with sdl 2.0.4 for the rpi on raspbian.

andyfrueh, 2016-03-08:
Boot screen shows Linux 4.1.13+

joolswills, 2016-03-08:
are you using a raspberry pi ? are you using Raspbian ? please can you post a log from RetroPie-Setup/logs/ after you get the failure (compressed - need to unpack and then post to pastebin.com)

joolswills, 2016-03-08:
do you definitely have a working internet connection also ?

andyfrueh, 2016-03-08:
I'm on the internet right now.  Same home connection.  I'm using a Pi 2B.  I've verified I can surf the web on the Pi.  The log is too large for pastebin unless I pay for an account.  Here's a pdf of the log.
[run_08032016_0739.pdf](https://github.com/RetroPie/RetroPie-Setup/files/163992/run_08032016_0739.pdf)

joolswills, 2016-03-08:
when you launch retropie-setup at the top of the main menu - does it say a version number ?

joolswills, 2016-03-08:
also it looks as though you are building from source ? I don't recommend that.

andyfrueh, 2016-03-08:
I'm choosing the first option from the retropie-setup script, "binaries."  There is no version information at all at the top of the screen.

joolswills, 2016-03-08:
then your retropie-setup script is not up to date. TBH if you are updating from a version that old, you should start from a fresh image - as we are dropping support for raspbian wheezy.

andyfrueh, 2016-03-08:
The 3.6 image is the latest, correct?  I'll back up all the ROMs and BIOS files.  Hope I remember to grab the INTV and GBA BIOS since they aren't kept in the same place as the others.  If that works, I suppose we can close this case.

joolswills, 2016-03-08:
ok np. note on the new image refer to the wiki as locations have probably changed (for the better) :)

closed 2016-03-08 by joolswills

HerbFargus, 2016-03-08:
Just follow this page and you'll be fine: 

https://github.com/RetroPie/RetroPie-Setup/wiki/first-installation

and as noted above see each emulator page on the wiki as many locations for things have changed

andyfrueh, 2016-03-08:
OK yeah, I see the emulator pages.  Got about half an hour to finish downloading ROMs then I'll flash a fresh card with the 3.6 image and see what that gets me.

andyfrueh, 2016-03-08:
Hmm...well, it burned the 3.6 image, and it booted, but when it tries to fire up ES, I get "Illegal Instruction" before the ES splash screen even loads.

andyfrueh, 2016-03-08:
Hmmm again...I may have grabbed the wrong version.  I was pretty sure I had a 2, but I only have one Raspberry on boot.  That means I have a 1 according to the wiki, right?

HerbFargus, 2016-03-08:
Bingo

joolswills, 2016-03-08:
Please continue on the forum for further help as we like to keep this issue tracker for tracking bugs only - https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md