RetroPie/RetroPie-Setup #1357: Russian Language

Snimchenko, 2016-03-08:
Hello,everyone!
I want to translate retropie interface to russian language.
How i can do it?

joolswills, 2016-03-08:
it isn't possible and we have no framework for localisation at all currently. apart from manually editing every string in the code. sorry.

Snimchenko, 2016-03-08:
> it isn't possible and we have no framework for localisation at all currently. apart from manually editing every string in the code. sorry.
Maybe you create a list of words that need to translate for me to i translated them?

joolswills, 2016-03-08:
there is so much on our todo list, we don't have a high priority for this sorry and I doubt we will do it. I believe recalbox has localisation though so you could try that.

joolswills, 2016-03-08:
even if we had all the strings translated, changing everything over to get translated would be more work than we have time for currently.

Snimchenko, 2016-03-08:
ok,thank you :)

closed 2016-03-11 by joolswills

MayMih, 2018-05-16:
It's strange so how in the retroOrangePi branch they achived localization support, at least in the EmulationStation menu?!