RetroPie/RetroPie-Setup #1360: Asks for controller configuration on each boot

PascalPencil, 2016-03-09:
After upgrading to the latest provided image of (only) RetroPie, the following happens:

1. 2 Logitech 710 wireless receivers are plugged in
2. These are both configured, however:
2. Each boot, the system reports detecting two controllers, and the hold a button on either of them to configure it
3. Pressing a single button on the controller 6 times gets rid of this popup

The system is used by children in my case and it's just anarchy in the house every time they want to play a game now

joolswills, 2016-03-09:
if you press a button once and waiting it should also be fine (it might need twice) - but the problem is they are not paired by the time ES loads so it doesn't now about them.  you can configure a keyboard on ES which will then never prompt for a controller if it doesn't find it, or put a delay in so your controllers are paired earlier.

PascalPencil, 2016-03-09:
The keyboard not being set up this time was indeed the problem, thank you so
much for taking the time to respond!

wrote:  

> if you press a button once and waiting it should also be fine (it might need
twice) - but the problem is they are not paired by the time ES loads so it
doesn't now about them. you can configure a keyboard on ES which will then
never prompt for a controller if it doesn't find it, or put a delay in so your
controllers are paired earlier.

GitHub](https://github.com/RetroPie/RetroPie-Setup/issues/1360#issuecomment-19
4268272).![](https://github.com/notifications/beacon/AC82HJbNFv-
7a0nSvQBQ07jWsFe2TtxGks5prrkggaJpZM4HskMZ.gif)

closed 2016-03-09 by PascalPencil

f0o, 2017-07-26:
Hi,

I'm experiencing the same behavior but I dont have a keyboard attached. Instead I have a working PS3 controller connected via Bluetooth.

I ran the setup multiple times and also placed the settings in /opt/retropie/configs/all/retroarch-joypads/ as per wiki.

Still on every boot i asks for the setup. It's very annoying.

Any pointers?