RetroPie/RetroPie-Setup #1369: Amstrad CPC Emulator: joystick emulation renders keyboard partially unusable

jammet, 2016-03-11:
Pressing s or x on the Keyboard also sends fire button actions. This should be disabled by default. The home computer encompasses a lot of different things, and many of them used through the keyboard, not a joystick.

joolswills, 2016-03-11:
which emulator ?

jammet, 2016-03-11:
Default choice. I'm not sure how to determine which is being run at the moment.

joolswills, 2016-03-11:
probably the libretro core, but it depends which version of retropie. check in this menu https://github.com/RetroPie/RetroPie-Setup/wiki/runcommand

joolswills, 2016-03-11:
Development for that is here - https://github.com/libretro/libretro-cap32

jammet, 2016-03-11:
I believe that very same emulator (I recognise the main menu) is used in Recalbox. When I tested that last, on a Raspberry Pi 2, this menu did not freeze then. Then again I know nothing much about the difference. I'm just reporting in the hope that it is something somebody finds interesting, and maybe can work with.

joolswills, 2016-03-11:
Did you choose AMSTRAD KEYBOARD in input controller setting ? see the documentation in the link above.

jammet, 2016-03-11:
Not yet. I will try that too. But I also found that the other cpc emulator,
called "capricepi" does not have the keyboard problem, and also does render
mode 2 in a resolution that works well for it. You cannot open the menu to
open disk files at all, but it has working hotkeys. Might be a good
compromise if I cannot fix the other emulator for this. :) Thank you.

closed 2016-03-11 by joolswills