RetroPie/RetroPie-Setup #1370: AMSTRAD CPC Emulation: "MODE 2" has bad resolution

jammet, 2016-03-11:
That mode is very pixellated regardless of relatively high host resolution.

joolswills, 2016-03-11:
we are not responsible for the emulator development btw - we ship with two emulators and I can point you to the upstream repository depending which one it is.

jammet, 2016-03-11:
lr-caprice32, is what I'm seeing at startup, so I supposed that is the one. It works fine, if all you ever want to do is play joystick based games. There is a lot more though, and even tools and other programs I would like to use in this, when I can. What is the other emulator?

joolswills, 2016-03-11:
regarding resolution - the emulator output will be upscaled. you can use scanline overlays or shaders, or bilinear filtering and adjust render resolution - please use the retropie forum for help with this. http://blog.petrockblock.com/forums/forum/retropie-project-forum/

joolswills, 2016-03-11:
see the documentation on the wiki https://github.com/RetroPie/RetroPie-Setup/wiki/Amstrad-CPC

closed 2016-03-11 by joolswills