RetroPie/RetroPie-Setup #1372: sound output mupen64plus

Pygmaee, 2016-03-12:
Hi, 
when switching the sound output from jack to HDMI or vice versa. Mupen64plus config in libretro config folder has to be changed by hand. Is there a elegant solution for fixing this?

joolswills, 2016-03-12:
This issue tracker is for bug reports please use the forum for discussion/support - https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md - I believe the audio plugin has been changed to SDL now which will not require manual configuration to switch between hdmi/jack - update retropie-setup and re-install from binary.

Pygmaee, 2016-03-13:
Thanks for your reply. i am new to github, but have quite some experience with linux, so i didn´t want to get support for get it working.
i wanted to help get it working out of box. I have the newest version installed, but i think SDL2 is only applied for input configuration. Audio output is Mupen64Plus OMX Audio Plugin.
An idea i have, is a script which changes the config file, but i think this isnt very elegant.

gizmo98, 2016-03-13:
There was a change after rp3.6 release. So if you update setup script and reinstall mupen64plus audio-sdl should be the default audio plugin.

closed 2016-03-13 by gizmo98