RetroPie/RetroPie-Setup #1376: Enhancement: Add-On or support for Phillips Hue lights

Twitch0815, 2016-03-15:
Phillips Hue has a great API.  

In Kodi there is an Addon, that pages 1-3 colors from the screen depending on how many bulbs you dedicate to it and then changes those bulbs to those colors.   So you just enable the addon and as you are watching a movie you get great ambient lighting.  

This would be awesome with gaming as well.  Could even integrate with Hyperion and Ambilight since they would pair well together I have this working in Kodi but would like to have it during the gaming experience.

joolswills, 2016-03-15:
I think this is somewhat out of the scope of RetroPie - perhaps you should see if anyone on the forum is interested or have a go yourself ?

Twitch0815, 2016-03-15:
I guess the intent would be to reverse engineer https://github.com/michaelrcarroll/script.kodi.hue.ambilight    and make it applicable to Emulation Station.

joolswills, 2016-03-15:
Might want to look at scripting this yourself - eg after googling I found http://www.developers.meethue.com/content/ambilight-hue-running-raspberry-pi

if you have a script that triggers it you could launch it before retropie etc.

joolswills, 2016-03-21:
This is really outside of the scope of the project, so am going to close. Thanks.

closed 2016-03-21 by joolswills