RetroPie/RetroPie-Setup #1377: Cannot use gamepad on EmulationStation (x86)

zefie, 2016-03-15:
I have tried both a PS4 controller and XBox 360 controller. The Xbox 360 controller is my primary, so I will address this in this post. I have set up the xboxdrv and permissions correctly, as the user that is running emulationstation can see the gamepad with jstest /dev/input/js0. EmulationStation claims to see the gamepad but when holding a button to configure it, nothing happens.

joolswills, 2016-03-15:
what are the contents of ~/.emulationstation/es_log.txt  (after trying to configure a controller)

joolswills, 2016-03-15:
also if you run emulationstation from a terminal - is anything outputted to the terminal ?

zefie, 2016-03-15:
I'm closing this because it is a configuration error I have yet to discover with how I'm initially launching emulationstation. If I launch it from a terminal in the desktop it works as intended, but if I use my current method of autostarting it, it does not, so this is my fault.

closed 2016-03-15 by zefie

gizmo98, 2016-03-18:
You can use retropie_setup.sh to configure autostart.

zefie, 2016-03-18:
retropie_setup.sh's method of autostarting does not work with 15.10, or at least did not for me on 64-bit, due to systemd rather than rc.d support. Same with xboxdrv. I did eventually figure out that I had to launch not only a terminal, but a window manager as well.

joolswills, 2016-03-18:
on ubuntu with Xorg desktop the autostarting uses a ".desktop" entry in ~/.config/autostart that should start a terminal with ES when you login to the desktop - unrelated to systemd / rc.d