RetroPie/RetroPie-Setup #1378: How to Change N64 Resolution

storyhobbit, 2016-03-15:
Hey guys!

I only have a small amount of technical expertise, but I've looked all over and can't find an answer to this question. I installed Emulation Station on my Raspberry Pi 2 (Model B), and it all works great, but I'm trying to change the resolution for the N64 emulator because it's far too sharp and all of the games have a lot of lag as a result. I don't need it to be more than 640x480, but for some reason the emulator renders it in HD. I've tested Playstation games too (since they require similar processing power), and they work fine at the appropriate resolution. Anyone know how to fix this?

HerbFargus, 2016-03-15:
Github is for bug reports. Please use the forum for support.

https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2016-03-15 by HerbFargus

storyhobbit, 2016-03-15:
Oh sorry, thank you!