RetroPie/RetroPie-Setup #1382: Berryboot with Rasp 2 & 3 support and Xbox 360 wireless remote not detected

Frucoboti, 2016-03-15:
Hello,
Any issue with the detection of the Xbox 360 wireless remote attached to a Rasp 2, with Berryboot for Rasp 2 only. Retropie works fine. It ask for press button if any remote is configured.
Once the configuration wizard ends, continues to Emulastation with no problem. The pad is fully operational.
In order to prepare the SD to use with a new Rasp 3, I have updated the Berryboot files at SD with new ones published a few days ago, maintaining the config.txt and cmdline.txt files.
Retropie 2.5 ask for joystick configuration, and don't detect any activity when push any button from same pad that worked with prior version.
Updated to Retropie 2.6 to check if this was the issue with no luck. Continues without detect anything.
Restored to original files and Retropie 2.5 and 2.6 continues with any issues detecting the remote, and without asking for configuration anytime, because detect original configuration.

Any idea how to solve this?

Thanks

zerojay, 2016-03-15:
Update to Retropie 3.6? 2.5 is really old now.

Frucoboti, 2016-03-15:
Sorry, I made a mistake. I meant always 3.5 & 3.6

I am going to edit post.

Thanks

joolswills, 2016-03-15:
I suspect they mean 3.5 / 3.6 - but if it works on our image, then it is likely related to Berryboot kernel etc. The berryboot images are provided as they were requested by a few people, but I cannot offer support for them really.

joolswills, 2016-03-15:
(you could ask on the forum to see if anyone else is using them and has an answer - which would be a better place than this issue tracker imho)

HerbFargus, 2016-03-22:
Tested on my berryboot build and it had no problem connecting to my xbox controllers so its likely something you've done on your end. as this is not a bug use the forum for further support.

https://github.com/RetroPie/RetroPie-Setup/blob/master/CONTRIBUTING.md

closed 2016-03-22 by HerbFargus