RetroPie/RetroPie-Setup #1383: Retroarch 1.3.0 problem

efraimsangil, 2016-03-16:
Hi!
I use Retropie 3.6.
This version uses Retroarch 1.3.0.

So, when I change any option in rgui, when I click on “Save New Config” appear this msg:

Failed saving config to “/opt/retropie/configs/snes/snes9x_next_libretro/.cfg”

The problem is that use “/” in filename -> snes9x_next_libretro/.cfg is filename.

Where I can fix that?

Regards

joolswills, 2016-03-16:
update retropie-setup and rebuild retroarch from source via menu 5 (or reinstall from binary as the binaries were updated recently), to see if it fixes it.

efraimsangil, 2016-03-16:
Downloading.... in a few minutes I tell you :) Regards

joolswills, 2016-03-16:
I think it isn't fixed, ill open a ticket on the retroarch bugtracker

efraimsangil, 2016-03-16:
I try it again and still fail...

joolswills, 2016-03-16:
https://github.com/libretro/RetroArch/issues/2793

MegaFish0527, 2016-03-18:
I have a RPi2 and this also happens to me, but slightly differently. The "Failed saving config to “/opt/retropie/configs/snes/snes9x_next_libretro/.cfg” error appears on the bottom left of the screen when opening a snes game. When I go into the RGUI and try to do something, it glitches and goes back to whatever the previous screen was. I updated it today (3/17/2016 5:00 PM).

lionking03, 2016-03-19:
the failed.... message wax fixed few hours ago, [https://github.com/libretro/RetroArch/issues/2793](url)
just rebuild from source retroarch and no message anymore
the issue wih the save config on RGUI are still there

joolswills, 2016-03-19:
Now fixed. I have updated the retroarch binaries also.

closed 2016-03-19 by joolswills

efraimsangil, 2016-03-19:
Ok, now it's working!!! Great job!