RetroPie/RetroPie-Setup #1385: RPI3 default compiler flags. neon-fp-armv8 has no effect if  -funsafe-math-optimizations is not specified.

joyrider3774, 2016-03-16:
RPI 3 flags are set to the following in system.h:
__default_cflags="-O2 -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard"

however specifying only -mfpu=neon-fp-armv8 does not have an effect if -funsafe-math-optimizations is not specified.

to quote from  https://gcc.gnu.org/onlinedocs/gcc-4.9.2/gcc/ARM-Options.html from the -mfpu section

> If the selected floating-point hardware includes the NEON extension (e.g. -mfpu=‘neon’), note that floating-point operations are not generated by GCC's auto-vectorization pass unless -funsafe-math-optimizations is also specified. This is because NEON hardware does not fully implement the IEEE 754 standard for floating-point arithmetic (in particular denormal values are treated as zero), so the use of NEON instructions may lead to a loss of precision. 

so i did a quick test using a whetstone benchmark from here http://www.roylongbottom.org.uk/Raspberry%20Pi%20Benchmarks.htm using default 4.9.2 pi compiler

1st test: not specifying  -funsafe-math-optimizations
cc whets.c cpuidc.c -lm -O3 -march=armv8-a -mfloat-abi=hard -mfpu=neon-fp-armv8 -funsafe-math-optimizations -o whetstonePiA8-neon-fp-armv8
`
Loop content                   Result              MFLOPS      MOPS   Seconds

N1 floating point      -1.12475013732910156       336.557               0.040
N2 floating point      -1.12274742126464844       328.799               0.290
N3 if then else         1.00000000000000000                1798.005     0.041
N4 fixed point         12.00000000000000000                1498.745     0.149
N5 sin,cos etc.         0.49911010265350342                  12.188     4.840
N6 floating point       0.99999982118606567       256.915               1.489
N7 assignments          3.00000000000000000                1198.953     0.109
N8 exp,sqrt etc.        0.75110864639282227                   8.712     3.028

MWIPS                                             710.041               9.985
`

2nd test: specifying -funsafe-math-optimizations 
cc whets.c cpuidc.c -lm -O3 -march=armv8-a -mfloat-abi=hard -mfpu=neon-fp-armv8 -funsafe-math-optimizations -o whetstonePiA8-neon-fp-armv8 -funsafe-math-optimizations
`
Loop content                   Result              MFLOPS      MOPS   Seconds

N1 floating point      -1.12367534637451172       518.495               0.027
N2 floating point      -1.12167274951934814       500.473               0.198
N3 if then else         1.00000000000000000                1798.265     0.042
N4 fixed point         12.00000000000000000                4495.306     0.052
N5 sin,cos etc.         0.49911010265350342                  12.165     5.047
N6 floating point       0.99999982118606567       256.801               1.550
N7 assignments          3.00000000000000000                1198.780     0.114
N8 exp,sqrt etc.        0.75110864639282227                   9.264     2.963

MWIPS                                             738.422               9.994
`
these tests were compiled on my pi3 itselve and ran while the cpu was NOT throttling due to tempratue settings the tests don't take long though but i am able to reproduce this fairly consitent specifying -funsafe-math-optimizations does seem to be needed for a further speed increase. It is also mentioned in the quote i posted above. 

When looking at the total MFLOPS there's an increase of 28000 with it specified however i have tried compiling ppsspp with that setting but did not *see* a real improvement

feel free to experiment / comment

joolswills, 2016-03-16:
I am aware of the option, but I am not sure that the text means that it will not be used at all - it doesn't say that - it says that neon floating point code is not generated by the auto-vectorization pass. It doesn't mean it is not used at all for floating point from my understanding.

Not sure that test proves too much (looks like close results) - certainly needs some more testing

I would add:

I did some additional benchmarks including quake (in timedemo) and it made no difference. Also no speed up from using the rpi2 compiler flags. Also note that a) those compiler flags are only used if building from source - rpi3 uses the rpi2 binaries.

many of the modules already specify -ffast-math which sets -funsafe-math-optimizations

joolswills, 2016-03-16:
Also in some cases on ARM -O3 can produce slower code - we don't use O3 by default either due to memory requirements for some code.

joolswills, 2016-03-16:
More data is needed on this certainly.

joolswills, 2016-03-16:
I re-read your table (would be clearer with some markdown :) ) and the first two tests def do better so it is worth considering. I do not want to maintain another set of binaries though unless there are some real world performance increases.

joyrider3774, 2016-03-16:
i added another test using -mfpu=vfp it produces for the first almost the same exact results as with -mfpu=neon-fp-armv8 without specifying -funsafe-math-optimizations. Thats why thought it is needed but one test probably does not prove this.

also added some indicators to the numbers that matter with asterixes.

but indeed i did not notice any speed diffrences in a "real" application though but i only tested ppsspp

joolswills, 2016-03-16:
Worth testing with -mfpu=neon-vfpv4 (rpi2 config) also.

joyrider3774, 2016-03-16:
add 2 more tests one with mfpu=neon-vfpv4 without -funsafe-math-optimizations and another with it specified.

i'm seeing the same increase.

But as stated before i have yet to find an emulator that benifits from it.

It just made me wonder about specifying -funsafe-math-optimizations or not

joolswills, 2016-03-16:
maybe we should just add it for rpi2 and rpi3 then - it shouldn't do any harm (I don't think anything relies on IEEE compliance but we will find out). We could consider adding -ffast-math - but there are also possibilities in some cases it could make performance worse (I have read about a case at least)

joolswills, 2016-03-16:
It's a shame it is showing zero improvement in quake as that uses a lot of floating point maths - I was just testing again with tyrquake (it has -ffast-math in the Makefile which disabled and then set flags manually with export CFLAGS=)

running tyr-quake -w -width 1920 -height 1080 -path ~/RetroPie/roms/ports/quake/id1/pak0.pak +timedemo demo3

with -O3 -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard -funsafe-math-optimizations

1090 frames  96.7 seconds  11.3 fps

with -O3 -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard

1090 frames  97.6 seconds  11.2 fps
```

note the values differ slightly on runs, but basically they are identical.

joyrider3774, 2016-03-17:
ah thats too bad, it seems it has no real impact on real applications well at least not in quake and ppsspp but i doubt we'll find one then. Weird that it did have an impact on the benchmark program. ah Well it was worth a shot, could have been it did had an impact, at least now we know it does not make that much diffrence

joolswills, 2016-04-09:
I think I will just include the flag for now, despite it only showing improvement in that benchmark. Perhaps something will benefit which doesn't already set this already

joolswills, 2016-04-09:
Thanks for all the feedback. I shall close this for now.

closed 2016-04-09 by joolswills

joolswills, 2016-10-17:
We have been building with `-O3 -funsafe-math-optimizations` for some months now, but looks as though this is causing issues in some cases - eg with lr-fba-next (some games crashing when O3 it is enabled).

I'm going back to `-O2`, and without  `-funsafe-math-optimizations`. 

If we find a case where real world performance is affected, we can include the optimisation on a per emulator basis, rather than system wide.