RetroPie/RetroPie-Setup #1386: asplashscreen

joolswills, 2016-03-16:
Look into splashscreen videos again (convert to systemd service?) - omxplayer sometimes starts before system is ready (error is - waiting for dbus address to appear). Need another solution for the hacky waiting for /tmp to be writable.

joolswills, 2016-05-10:
Although some other users reported this here

http://blog.petrockblock.com/forums/topic/splash-video-not-loading-anymore/

I am currently unable to reproduce. Video is showing fine each time.

joolswills, 2016-05-10:
Scrap that - I did just get it to fail.

joolswills, 2016-05-10:
So I think what was happening:

We were incorrectly waiting for /tmp to be writable (which seemed to work before), and then starting asplashscreen. We really just wanted to make sure dbus is running as else omxplayer complains. On some boots with the wait for /tmp dbus wasn't actually running - which caused omxplayer to start outputting errors. I believe this triggers systemd to disable the init.d script from startup since it doesn't finish in a required amount of time.

We now wait instead for the "dbus" process. I have rebooted lots of times now, and it still seems to be showing. I also had to re-enable the splashscreen service for it to start working again (via retropie-setup)

closed 2016-05-10 by joolswills

HerbFargus, 2016-05-10:
I can confirm that it is working on my end as well.

commit f13b3ea5 (2016-05-10): wait for dbus before starting omxplayer - hopefully fixes #1386