RetroPie/RetroPie-Setup #1387: Raspi3: Emulationstation crash on warm reboot

andrewsil1, 2016-03-18:
This is probably an upstream bug, but I've found a 90%-ish reproducible crash in the following circumstances:
- Raspi 3. Not sure if this is relevant, might well happen on a Raspi 2 but I'm not certain.
- Emulation station has the Carbon theme installed and system sounds are turned ON for the menus.
- Sound default device is snd_usb_audio (C-media Headphone Set from Adafruit)

After a -warm- reboot, but not a cold one, EmulationStation will make some ugly noises through the speakers and then crash, leaving the display in the menu mode.  I suspect it's dropped to the terminal behind it, but you can't see it on screen.  On a cold boot, everything works fine, however.

joolswills, 2016-03-18:
Well it sounds related to that hardware - but it isn't something we are going to be able to fix. Also this isn't the emulationstation repository. How are you powering your RPI ? What PSU ?

joolswills, 2016-03-18:
also If it works on cold boot but not warm, it sounds like a hardware compatibility or driver issue.

andrewsil1, 2016-03-18:
I figured as much, but at least thought I'd ask. :-)  I'm powering the RPi with an industrial power supply capable of up to 20A, so that shouldn't be the problem. :-)

The only thing that makes me think it might be ES is that with this audio device it always complains about not being able to find the mixer elements (because there isn't one, I think), but I've had that bug open on them for > 1 year, and I don't think much is happening.

closed 2016-03-18 by andrewsil1

joolswills, 2016-03-18:
if it is fine on a cold boot I doubt it is ES - I guess the hardware is initialised differently. You could perhaps tinker with alsa / driver settings also.