RetroPie/RetroPie-Setup #1388: rpi3 and nimbus

trontech83, 2016-03-19:
I have a SteelSeries Nimbus and, using bluetoothctl, it connects.  After reboot, it will connect just fine. However, it will not be recognized in the gamepad configuration.
I've looked all over but most refer to the PlayStation controller or NES (8btdio)  bluetooth.
Please help me get my paired and trusted BT controller recognized by RetroPie! :)

I also have a MadCatz C.T.R.L.i
it will not pair but I'd love to get this working as p2

Thanks

joolswills, 2016-03-19:
Maybe the controller isn't compatible with linux / bluez. You can see if you get anything from jstest, but you may have to resort to another controller - see if you can find anyone else using it on linux perhaps?

trontech83, 2016-03-19:
oh, honestly, I didn't realize "isn't compatible" was an option :/
I'll look into it, Thank you

trontech83, 2016-03-19:
should I be using Bluez? I hear it's unstable; something to do with the GUI aspect...
When I check devices using the built-in BT in the menu it shows connected :/

joolswills, 2016-03-19:
bluez is the linux bluetooth stack so you are already using it - you are thinking about it's desktop GUI perhaps.

trontech83, 2016-03-19:
yes, I think I am.  After expand, rpi-update, update, upgrade and retropie install I get pi-bluetooth but not bluez. I do not get a gui but are they the same? Working on another project ATM but will get a ps3 controller for testing. Thanks  

If I'm paired with bluetoothctl does that pretty much mean it's a compatibility issue??

joolswills, 2016-03-19:
bluez is the bluetooth stack. bluetoothctl etc are all part of bluez. For further help please use the forum - you might be able to find others with a similar device. You should get info in "dmesg" regarding the device and you can also check for /dev/input/js0 (and use jstest /dev/input/js0 )

closed 2016-03-19 by joolswills

JulyIGHOR, 2018-05-26:
Its recognized as HID device and need driver to make it works.
SteelSeries Nimbus is sending 18 numbers, each one is responsible for different button / axis. And each button recognize pressure so it can be used as axis.
Here is Windows driver https://github.com/Axadiw/MFIGamepadFeeder that makes more better understanding how it works.
All MFi gamepad controllers works same way, so driver will be universal.