RetroPie/RetroPie-Setup #1392: Failed to boot with new experimental OpenGL driver

yangxuan8282, 2016-03-19:
I am on RPI3,  RetroPie 3.6. 
After enable GL Driver, 
in *raspi-config*, under Advanced Options->GL Driver.
RetroPie won't boot, display coloured splash screen [like this](http://elinux.org/File:Debug-screen.jpg) .

HerbFargus, 2016-03-19:
Of course it won't boot. The opengl driver only works with a desktop environment RetroPie doesn't ship with.

closed 2016-03-19 by HerbFargus

AlgorithMan-de, 2017-10-06:
And how do you turn the driver off when you only get a console, but keyboards aren't activated and SSH is off by default?

AlgorithMan-de, 2017-10-06:
Also I want to add that it is a terrible UX design to have a MENU-OPTION that breaks the whole OS.

AlgorithMan-de, 2017-10-06:
I edited the file config.txt on the boot partition and comment out the line
dtoverlay=vc4-fkms-v3d
That at least brought back some graphics mode, but I only got an EmulationStation without background and a menu with only empty lines and nothing worked. I have flashed the SD-Card anew now.

Thank you very much for wasting me 2 hours, whoever had the idea to hide a "brick the OS" function in a "this sounds like a very good idea" menu entry.

PokeEngineer, 2017-11-22:
@AlgorithMan-de 

The following might seem off-topic at first, but gives background for later. Alright, here we go:

RetroPie is not an OS, it is a shell script system that runs off an OS called Raspbian. Raspbian was made by the Raspberry Pi Foundation, not by us. Raspi-config was also not made by us, but was made the same people who made Raspbian and was preloaded in there. As such, what is in that menu is not in our control. So, the OpenGL driver was meant for the regular desktop version of Raspbian and not the modified Raspbian Lite image that is RetroPie. So sorry, we can't do anything about this...