RetroPie/RetroPie-Setup #1394: RPi3 unable to update Retropie script

LodanZark, 2016-03-21:
trying to update the Retropie script (on retropie 3.6) gives this error:

Update Failed
fatal: unable to access
'https://github.com/RetroPie/RetroPie-Setup.git/':
server certificate validation failed. CAfile:
/etc/ssl/certs/ca-certificates.crt CRLfile: none

to fix this situation i needed to apply this command:
git config --global http.sslverify false

maybe this configuration should be set up on sd image  to avoid this problem

joolswills, 2016-03-21:
This is not a problem on the image as far as I know - have you removed the ca-certificates package?

LodanZark, 2016-03-21:
No I didn't, should I?

joolswills, 2016-03-21:
No - it's needed. Your error is either the package was removed and needs re-installing or you have d card corruption.

joolswills, 2016-03-21:
(It's not a retropie-setup bug anyway)

LodanZark, 2016-03-21:
was a fresh install with a fresh new sd card (16gb kingston class10), before I try update the retropie script I just updated the OS over sudo apt-get update && sudo apt-get upgrade that's all, maybe it did update some packages related with ca-certificates packages as result, I don't know. So the solution is reinstalling the ca-certificates?

joolswills, 2016-03-21:
if the problem is due to that - you can try and reinstall.

LodanZark, 2016-03-21:
thank you, i will give a try

closed 2016-03-21 by joolswills

torrancew, 2016-05-03:
FYI, I hit this same issue after installing some things manually via apt. I was able to work-around it by issuing the following command inside of `/etc/ssl/certs`:

```
$ sudo c_rehash .
```

Where `c_rehash` is Raspbian's name for the `openssl rehash` command.

This suggests that this directory is being managed incorrectly during the RetroPie install process (Normally you would add certificates to the appropriate location under `/usr`, and run the distro-provided helper script to install the proper symlinks into `/etc/ssl/certs`), though I've not yet had time to investigate. I'm a bit puzzled/confused as to what the certs mismatch was, as I'd normally expect Retropie to embed a CA certificate in the system trust root.

joolswills, 2016-05-03:
I dont think there is any problem on the premade image afaik - but there are various reasons to have errors with certs. I think the most common one is the system time being incorrect, but I have tested with a fresh image and all was fine.

torrancew, 2016-05-04:
This is most certainly *not* a time issue. Consider the following:

  * `curl https://rubygems.org` fails with SSL errors
  * `curl https://google.com` does **not** fail
  * `curl --cacert /etc/ssl/certs/ca-certificates.crt https://rubygems.org` does **not** fail
  * `curl --cacert /etc/ssl/certs/ca-certificates.crt https://www.google.com` does **not** fail
  * After running the `sudo c_rehash /etc/ssl/certs`, all commands that previously failed, do **not** fail
  * When not specifying `--cacert`, `curl -v` reports that it is attempting to use the **directory** `/etc/ssl/certs` as its CA path (aka `--capath` from the CLI)

I noticed this while attempting to install puppet from a gem. The only actions taken prior to this were installing Kodi from the RetroPie UI, and installing `salt` from apt. I have not reflashed to test a vanilla image for its functionality against `https://rubygems.org`, which is using a CA that was impacted by whatever the RCE here is.

joolswills, 2016-05-04:
There have been cases that have been time issues though (on the forum) which is why I suggested it. I can reproduce at least one of the above problems though on an image, so will make sure the next image doesn't have the issue. No sure what may have caused it - but will make sure the next image passes the above. Can only think it was caused by some package update or something.

joolswills, 2016-05-04:
I removed/re-installed the certificates package already on my dev chroot, so I will just test that works with the above - I suspect it is already fixed.

torrancew, 2016-05-04:
Indeed, I'm all too familiar with many of the common TLS troubleshooting tasks. Thank you for looking into this!

joolswills, 2016-05-04:
chroot seems ok, so looks like it is already fixed from when I reinstalled the certs package the other day (when someone mentioned a problem with kodi). Thanks for the feedback.

renevn, 2016-05-12:
Adjust your raspberry date and time, this can be making the certificates to not work properly. I had this problem and solved just by sudo date .....

torrancew, 2016-05-12:
@renevn Please see my above comments. This specific issue is not date-related, though I appreciate the advice.

renevn, 2016-05-12:
Did not see your comments about date before.. unfortunately not the same issue..

1031982, 2016-08-06:
I had the dame problem on a new Pi3 setup. Doing a manual install of RetroPie had no issues.