RetroPie/RetroPie-Setup #1395: RetroPie Menu Icons

HerbFargus, 2016-03-21:
Rookervik and I have discussed options for icons in the RetroPie menu of EmulationStation. As EmulationStation wasn't designed with a settings menu in mind there is no simple way to switch themes  with preloaded boxart with preloaded gamelists without either switching a boxart forder or gamelist.xml file.

As a compromise Rookervik has modeled icons in blender that should be sufficient for mutliple themes if switched. I have added them to the RetroPie-Menu branch of the Carbon Theme (https://github.com/RetroPie/es-theme-carbon/tree/RetroPie-Menu)

The esthemes module will need a bit of tweaking to install the gamelists and artwork in `/home/pi/retropiemenu` or if we don't want to do that we can just ship with the gamelists and icons as part of the image. I don't feel its worth our time to hardcode in other theme's gamelists due to the aforementioned theme switching issues as above so it will be left to the user to manually add their own menu artwork for their respective themes if they want different menu artwork.

(From the EmulationStation documentation artwork can go in 3 places:

- `[SYSTEM_PATH]/gamelist.xml`
- `~/.emulationstation/gamelists/[SYSTEM_NAME]/gamelist.xml`
- `/etc/emulationstation/gamelists/[SYSTEM_NAME]/gamelist.xml`

The format I've chosen is to have an `icons` folder in the `/home/pi/retropiemenu` and then subfolders for each theme are placed in there which means people will only have to switch the gamelists if they switch themes.  so in icons there will be a folder for carbon, nbba, etc, and gamelists will just point to different image folders depending on the theme that is chose.

joolswills, 2016-03-21:
Thanks - probably would have been best to do a PR first so I can give some feedback.

do you mean ~/RetroPie/retropiemenu btw ?

This folder might be moved btw - Was going to move it to /opt/retropie/configs/all/ 

We don't want to include the gamelist in the theme with the hardcoded paths - I need to be generated by the script. RetroPie-Setup can be installed by any user.

joolswills, 2016-03-21:
Please can you revert this until I have had a chance to look at the best way of implementing it.

HerbFargus, 2016-03-21:
yeah `~/RetroPie/retropiemenu` is what I meant. but yeah I put it on a branch because I knew that there would need to be some discussion on the matter.

joolswills, 2016-03-21:
sorry my bad. Didn't see that.

joolswills, 2016-03-21:
Thought it was on master branch :)

HerbFargus, 2016-03-21:
I have some screenshots on my other computer that I'll add once I get access to it again.

joolswills, 2016-03-21:
I think it's fair enough to just include this in carbon etc - but if other themes wanted it implement it they could I guess? I will move the gameslist generation to the script for now and we can go from there.

HerbFargus, 2016-03-21:
Rookervik did make a pixel based version and NBBA I believe has their own retropie menu icons as well, and its possible others may wish to implement something similar in the future. I know that @InsecureSpike has done a lot with his themes and may possibly be interested as well.

joolswills, 2016-03-21:
I'm not sure I completely understand the theme switching problem and why we would need subfolders for each theme - but will have a look at the xml.

HerbFargus, 2016-03-21:
Menu boxart isn't specific to themes so when you switch a theme it won't carry over so you'll still see the old theme's images on the new theme you just switched to.

joolswills, 2016-03-21:
so the same icons will be used - but that's ok I think.

HerbFargus, 2016-03-21:
yeah that's fine, but if other themes like nbba have their own icons people will have to manually change  their gamelist xml to point to their icons instead of the default icons. which I'm fine with letting people set up other themes manually, I'm primarily concerned with Carbon as the default.

joolswills, 2016-03-21:
it could be incorporated into the theme installer I guess, but I'm not sure it's worth the effort - also the settings may be switched around as we see fit, so other themes should just use our icons and not worry about it.

HerbFargus, 2016-03-21:
yeah personally I don't think its worth the effort. I think a universal icon option is best, those that want something different can set it up manually (as they already do)

closed 2016-04-09 by joolswills

InsecureSpike, 2016-04-09:
sadly, i'dve like to have seen these as theme dependant, i have 2 reasons for this, 
1: i would disable them in my theme (i don't really see the need for them)
2: they could be designed to fit in with the creators theme, style wise

joolswills, 2016-04-09:
You are a little late - the ticket was discussed 19 days ago.

I guess you may be able to do something about it in your theme, but if you read above you will see there is no easy solution for a per theme style.

commit 0de788de (2016-04-09): added new gamelist.xml with descriptions and icons for retropiemenu.
removed dispmanx configuration as most users don't need to use it
big thanks to Rookervik and Herb
fixes #1395