RetroPie/RetroPie-Setup #1396: RetroPie 3.6 Filesystem resize

Konoa, 2016-03-22:
since getting my raspberry pi 3 i put retropi 3.6 on a 32gb sd using pi baker app. and it sits on the "first boot - Resizing filesystem...".

i then tried the same retropi 3.6 on my Raspberry pi 2 and it does the exact same thing

any help is greatly appreciated!

joolswills, 2016-03-22:
Please give the steps you went through in the app - Are you trying to create a noobs image ?

joolswills, 2016-03-22:
Also please link to the software you are using - I am not aware of it and the one I found when googling may be the wrong one.

Konoa, 2016-03-22:
I'm using ApplePie Baker on my macbook pro.
I've used retropie 3.5 and 3.6. and thats with restoring to those img on blank sd.

joolswills, 2016-03-22:
It would have been easier if you had pasted me a link as asked. I also asked "Please give the steps you went through in the app - Are you trying to create a noobs image ? "

Konoa, 2016-03-22:
format sd to fat32
click restore backup
select either 3.5 or 3.6 retropi img
it does its thing
then i click eject
put it in pi
then pi starts doing its thing and comes to the expanding filesystem (which is auto in retropie 3)
and it just hangs there on that screen

this is the app http://www.tweaking4all.com/hardware/raspberry-pi/macosx-apple-pi-baker/

joolswills, 2016-03-22:
There shouldn't be any need to format it to fat32 first.

if you can mount the image/microsd on the mac you can remove the resize script which is in /etc/profile.d/ but the script calls the standard raspi-config so not sure what is going wrong, but I suggest first trying to manually write the image to see if it makes any difference. 

https://www.raspberrypi.org/documentation/installation/installing-images/mac.md

no-one else has reported this problem - the only other known issue with the resize is when converting the image to noobs - which is documented on the wiki (https://github.com/RetroPie/RetroPie-Setup/wiki/Convert-RetroPie-SD-Card-Image-to-NOOBS-Image)

joolswills, 2016-03-22:
I also don't have OSX, so can unable to test this specific app - but you could ask on the RetroPie forum to see if anyone else uses it.

dankcushions, 2016-03-22:
I use OSX and used the 'command line' method on https://www.raspberrypi.org/documentation/installation/installing-images/mac.md to write 3.6 to an SD card, with no issues.

I suggest @Konoa uses the 'command line' method to try again and then if it works then we know that the issue is with this 'apple pi baker' tool.

InsecureSpike, 2016-03-22:
all works ok for me using the latest img with apple pie baker, but I'm using a 4 or 8gb card, maybe it's an issue with larger cards?

craigerskine, 2016-03-22:
I use [Pi Filler](http://ivanx.com/raspberrypi/) and it works awesome. Just did a new img of 3.6 yesterday and it was fast and simple. I've been using Pi Filler since the 2.X days of RetroPie.

[Pi Copier](http://ivanx.com/raspberrypi/) is also very useful.

joolswills, 2016-03-22:
Going to close the ticket as it's not a retropie-setup issue as far as I can see (and if there is a compatibility issue between our resizing and this tool, raspi-config resize would fail too as we use its resize function)

closed 2016-03-22 by joolswills