RetroPie/RetroPie-Setup #1399: [Feature Request] Default support for NESPAD by Xgaming?

Hedda, 2016-03-22:
Please consider adding default support for the "NESPAD", which is a new cross-platform gamepad in extended SNES/NES/N64 retro-style combo by Xgaming of X-Arcade fame.

http://shop.xgaming.com/products/nes-wireless-gamepad-ios-android-mac

I know that Shawn Walters <shawn@xgaming.com> from Xgaming donated an Tankstick (XArcade) to developers before and bet they donate NESPADs too if asked ;)

joolswills, 2016-03-22:
Looks like http://www.8bitdo.com/nes30pro/ which is already supported. if it works as a normal bluetooth joystick it should work and doesn't need "supporting". Although there are somethings to note when using 8bitdo controllers (documented on the wiki)

closed 2016-03-22 by joolswills

Hedda, 2016-03-23:
Please note that this case be used in wired (USB) mode too.