RetroPie/RetroPie-Setup #1402: Improvement: ScummVM 1.8.1pre config file -> config directory

jimbo27, 2016-03-23:
Recently I installed the new version of ScummVM over my existend version via RetroPie. After fiddling around with it I found out:
The config file $home/.scummvmrc is no longer used in v1.8.1pre. Instead there is the new config file $home/.config/scummvm/scummvm.ini

I suggest to abandon the symbollink for .scummvmrc and instead use this new one:
ln -snf $configdir/scummvm $home/.config/scummvm

I tested this on my system. It works flawlessly.

joolswills, 2016-03-23:
that folder is already symlinked I believe so the creation of the other symlink can just be removed.

joolswills, 2016-03-23:
aah no sorry my bad we symlinked the /home/pi/.scummvm folder - will update.

joolswills, 2016-03-23:
fixed.

closed 2016-03-23 by joolswills