RetroPie/RetroPie-Setup #1403: SEGFAULT with mupen64plus-GLideN64

dankcushions, 2016-03-23:
NOTE this only happened for me when I rebuilt from source - it did work before then, but I'm not sure how?

The cause of this is that GLideN64 is trying to log the following error message:
https://github.com/gonetz/GLideN64/blob/8d3e6ef2f9d8bb05b2d3670bc7d0842aedf0fedd/src/OpenGL.cpp#L1378
It wants to create gliden64.log in /opt/retropie/emulators/mupen64plus/lib/mupen64plus/, but fails as it does not have permissions. Here is the segfault:
````
pi@retropie:/opt/retropie/emulators/mupen64plus/lib/mupen64plus $ gdb --args /opt/retropie/emulators/mupen64plus/bin/mupen64plus --noosd --fullscreen --gfx mupen64plus-video-GLideN64.so --configdir "/opt/retropie/configs/n64" --datadir "/opt/retropie/configs/n64" "/home/pi/RetroPie/roms/n64/Super Mario 64 (USA).n64"
GNU gdb (Raspbian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/retropie/emulators/mupen64plus/bin/mupen64plus...(no debugging symbols found)...done.
(gdb) tty /dev/tty1
(gdb) run
Starting program: /opt/retropie/emulators/mupen64plus/bin/mupen64plus --noosd --fullscreen --gfx mupen64plus-video-GLideN64.so --configdir /opt/retropie/configs/n64 --datadir /opt/retropie/configs/n64 /home/pi/RetroPie/roms/n64/Super\ Mario\ 64\ \(USA\).n64
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x7268b460 (LWP 4949)]
[New Thread 0x70e89460 (LWP 4950)]
[New Thread 0x704ff460 (LWP 4951)]
[New Thread 0x6fcff460 (LWP 4952)]
[New Thread 0x6f4ff460 (LWP 4953)]

Program received signal SIGSEGV, Segmentation fault.
_IO_vfprintf_internal (s=0x0, 
    format=0x71a4b1b0 "N64 depth compare and depth based fog will not work without Image Textures support provided in OpenGL >= 4.3 or GLES >= 3.1", ap=...)
    at vfprintf.c:1278
1278	vfprintf.c: No such file or directory.
(gdb) bt
#0  _IO_vfprintf_internal (s=0x0, 
    format=0x71a4b1b0 "N64 depth compare and depth based fog will not work without Image Textures support provided in OpenGL >= 4.3 or GLES >= 3.1", ap=...)
    at vfprintf.c:1278
#1  0x7197bdb8 in LOG(unsigned short, char const*, ...) ()
   from ./mupen64plus-video-GLideN64.so
#2  0x7198a9c8 in OGLRender::_initExtensions() ()
   from ./mupen64plus-video-GLideN64.so
#3  0x7198b10c in OGLRender::_initData() () from ./mupen64plus-video-GLideN64.so
#4  0x7198b1f4 in OGLVideo::start() () from ./mupen64plus-video-GLideN64.so
#5  0x719965c8 in RomOpen () from ./mupen64plus-video-GLideN64.so
Cannot access memory at address 0x800000
#6  0x72868bb8 in ?? ()
   from /opt/retropie/emulators/mupen64plus/lib/libmupen64plus.so.2
Cannot access memory at address 0x800000
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
````

If you CHMOD this directory to 777, it creates the log fine, and proceeds without issue.

I don't think the log tolerance can be changed - I think it always wants to log warnings:
https://github.com/gonetz/GLideN64/blob/438f5e327b94302b97493c4cf84130ca265bcc58/src/Log.h#L11

I guess there are a number of ways this could be fixed
- fiddle the default permissions
- stop the error being logged by turning off these graphics options in the .cfg file (not sure if this is possible)
- redirecting the logfile via symlink?
- etc.

Not sure of the best approach, so thought I'd flag it :) It was also raised on the forum: 
http://blog.petrockblock.com/forums/topic/gliden64-plugin-will-not-run-rooms-all-other-plugins-work/
(these users don't appear to have rebuilt from source)

dankcushions, 2016-03-23:
Looks like this recent commit would have caused our issues:
https://github.com/gonetz/GLideN64/commit/438f5e327b94302b97493c4cf84130ca265bcc58

dankcushions, 2016-03-23:
I guess gliden64 shouldn't really crash just because it doesn't have permissions to create a logfile...

joolswills, 2016-03-23:
https://github.com/gonetz/GLideN64/blob/438f5e327b94302b97493c4cf84130ca265bcc58/src/Log.h#L29 (doesn't do any checks)

joolswills, 2016-03-23:
This should be reported upstream

joolswills, 2016-03-23:
To work around it in the meantime, we could symlink it as suggested (moveConfigFile will do the job) to the configs area.

dankcushions, 2016-03-24:
Logged upstream issue: https://github.com/gonetz/GLideN64/issues/930

dankcushions, 2016-03-24:
Solved by @joolswills upstream, thanks!

closed 2016-03-24 by dankcushions