RetroPie/RetroPie-Setup #1405: Raspberry Pi freezes immediately after finishing scrapping with error msg

Seppo007, 2016-03-26:
Directly after ending the scraping-process inside Emulationstation the pi freezes.
Emulationstation aborts with the following error message:

emulationstation: /home/pi/RetroPie-Setup/tmp/build/emulationstation/es-core/erc/components/IList.h:124: const UserData& IList<EntryData, UserData>::getSelected() const [with EntryData = TextListData; UserData = FileData*]: Assertion 'size() > 0' failed.
Aborted

I can't acces the IList Header myself, it seems that the header is generated when the Scraper is starting to scrape.

The problem always occurs independently of the systems that are selected to scrape for and the scraping provider.

Also the integrated WLAN-module seems to be turned off or the whole system got catched up in an inifinite loop, because you even can't access the pi via SSH to reboot it.

So the only option is to hard reset the pi via unplugging the pi from the socket.
Had the same problem with Raspbian Wheezy and Pi2 now using Retropie 3.6 on Raspbian Jessie with Pi 3.

joolswills, 2016-03-26:
Our emulationstation issue tracker is here - https://github.com/RetroPie/EmulationStation however we are not planning on doing any fixes for its scraper - recommend you use the scraper module instead (which is significantly faster too) https://github.com/RetroPie/RetroPie-Setup/wiki/scraper

Seppo007, 2016-03-26:
Thank you for your hint, I posted it right away in the corresponding issue tracker.

I knew that there was another Scraper implemented in the Retropie-Setup menu, but since i started with Retropie some years ago i always used the scraper inside the emulationstation-menu. Didn't know that this is even faster, i will give it a try.

joolswills, 2016-03-26:
your rpi3 wifi is probably going into power saving mode - there are tickets about this on the forum. doing `sudo apt-get update; sudo apt-get dist-upgrade` will get a recent kernel with power saving disabled - note we are not responsible for the underlying OS.

Seppo007, 2016-03-26:
My Pi is always up to date, i also updated the firmware to 4.4.3 to enable channels 12 and 13 for wifi connection. I will have a look for those issues with power saving mode

closed 2016-03-26 by Seppo007

joolswills, 2016-03-26:
The ES issue is unlikely to be looked at anytime soon (and it may well be triggered by a file on your system - i've not seen this problem myself and I have used the scraper recently).