RetroPie/RetroPie-Setup #1407: Reicast multiple controller mapping broken

vretro, 2016-03-28:
Auto controller mapping for Reicast is broken for more than 1 controller.

Only player 1 controller is recognised in game (any).
Tested using 2 Buffalo SNES controllers and only 1 is detected and mapped.

Controllers both setup using EmulationStation before launching clean installs of Reicast.

joolswills, 2016-03-28:
Our configuration system only currently handles the configuration of a single controller currently.

vretro, 2016-04-01:
Possible solution for multiple controllers in Reicast: http://blog.petrockblock.com/forums/topic/dreamcast/page/2/#post-121888

vretro, 2016-04-06:
Another possible solution for multiple controllers in Reicast:
http://blog.petrockblock.com/forums/topic/dreamcast/page/2/#post-122354

gizmo98, 2016-06-19:
Multiple controller support is possible now.

closed 2016-06-19 by gizmo98