RetroPie/RetroPie-Setup #1410: Nyko Playpad support

hanthecat, 2016-03-30:
I have a Nyko Playpad bluetooth controller. It syncs up just fine and I see most buttons work under jstest except for what should be the select button. Emulationstation only sees that select button as a keyboard press and ignores the rest.

Also, Im running fbsp to get my waveshare screen working. You can see that behind the framebuffer copy the console is still active. I get the text "They key you just pressed is not recognized by SDL" trying to force itself over the fb video.

Also, I have a ps3 controller that works great though that is not as ideal for travel as the Nyko Playpad.

Can someone point me in the right direction to get this guy working?

Thank you!

HerbFargus, 2016-03-30:
You already have thread open on the forum - please wait for a reply there.

closed 2016-03-30 by HerbFargus

hanthecat, 2016-03-30:
I understand.
I posted here as someone tried to ask the same question here and did get a reply
(https://github.com/RetroPie/RetroPie-Setup/issues/1173) though that thread closed as he didnt follow up.
There was another thread on the forum that didnt get a reply.Rather than reopen a thread that is months old I started a new one with more information as to what I have been able to troubleshoot so far.

I was trying to maximize the number of eyes that saw this request to hopefully get some direction on how to resolve. I apologize if you see it as redundant.

Thank you.