RetroPie/RetroPie-Setup #1411: SHANWAN PS3 GamePad

guancio, 2016-03-30:
Hi,
I have a SHANWAN PS3 GamePad. I'm failing to pair the controller with RetroPi.
I've installed the SHANWAN PS3 GamePad support via the setup-script, however, after
disconnecting the USB cable, the controller do not pair with the bluetooth controller.

I've notice some strange facts:
1) when I connect the controller via the USB cable it starts to vibrate and never stop (until I do not remove the cable)
2) when I execute sudo sixpair I obtain the following output:
Current Bluetooth master: ff:00:03:8f:9f:7b
Setting master bd_addr to 00:1a:7d:da:71:04
However if I execute the same command again, the output does not change. Does it means that the
current Bluetooth master is not updated?

My dmesg:
[  611.213027] usb 1-1.5.4.2: new full-speed USB device number 10 using dwc_otg
[  611.315843] usb 1-1.5.4.2: New USB device found, idVendor=054c, idProduct=0268
[  611.315874] usb 1-1.5.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  611.315927] usb 1-1.5.4.2: Product: PS3 GamePad
[  611.315943] usb 1-1.5.4.2: Manufacturer: SHANWAN
[  611.533062] usb 1-1.5.4.2: reset full-speed USB device number 10 using dwc_otg
[  611.661390] input: SHANWAN PS3 GamePad as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5.4/1-1.5.4.2/1-1.5.4.2:1.0/0003:054C:0268.0004/input/input2
[  611.662171] sony 0003:054C:0268.0004: input,hiddev0,hidraw2: USB HID v1.10 Joystick [SHANWAN PS3 GamePad] on usb-3f980000.usb-1.5.4.2/input0

joolswills, 2016-04-09:
Please use the forum first for help with controllers https://retropie.org.uk/forum/

closed 2016-04-09 by joolswills

crazyazz, 2016-11-09:
Did you find a solution to this issue?

BruneXX, 2017-04-29:
Hey, same problem here, did you find a solution to that issue?

pablomacaluso, 2017-11-13:
Same problem .

joolswills, 2017-11-14:
Please use the forum for help.