RetroPie/RetroPie-Setup #1417: RetroPie Controls - enable up/down + left/right

abb122, 2016-04-06:
I just set up my Raspberry Pi 3 with RetroPie so I can play some old school (NES & SNES) games, but something is off about the controls.  I wouldn't call it lag, but there's some sort of an input problem when I try to make sudden movements in games like Mario.  I'm thinking it has to do with pressing buttons together, for example suddenly changing direction from left to right.  From my experience using NESTopia on Windows, there's an option in the input settings to enable "up/down + left/right" simultaneously.  I'm thinking that this is the problem.  Is there a way to enable this in RetroPie?  Or could there be another cause to this issue?

Twitch0815, 2016-04-06:
Try changing the below in your retroarch.cfg file. 

# Attempts to hard-synchronize CPU and GPU. Can reduce latency at cost of performance.
video_hard_sync = true

# Sets how many frames CPU can run ahead of GPU when using video_hard_sync.
# Maximum is 3.
 video_hard_sync_frames = 3

# Sets how many milliseconds to delay after VSync before running the core.
# Can reduce latency at cost of higher risk of stuttering.
# Maximum is 15.
video_frame_delay = 5

joolswills, 2016-04-06:
note that `video_hard_sync = true` and `video_hard_sync_frames` are not used on the RPI.

joolswills, 2016-04-06:
This would be best posted to our new forum first - https://retropie.org.uk/forum/ rather than the issue tracker as it's likely not a RetroPie bug - but best to get some help on the forum first either way.

joolswills, 2016-04-09:
Closing - please use the forum for further help with this https://retropie.org.uk/forum/

closed 2016-04-09 by joolswills