RetroPie/RetroPie-Setup #1420: PS3 controller (wired) won't register button presses

rsrobinson87, 2016-04-08:
I have tried multiple times now to configure my PS3 controller both when I boot up retropie/emulationstation as well as the automatic configuration through the emalutionstation config menu as instructed by the directions here at GitHub.

Every time i try to map the buttons in the configuration menu, it allows me to hold a button and correctly identifies that it is a PS3 controller, but it does not register or react at all when i press the buttons on the controller to map them. For example, the first option says "D-pad Up" and when i press the d-pad up button nothing happens. The only buttons that have worked (and not at the right place) has been the circle button and sometimes the analog sticks and trigger buttons (and even that is hit-or-miss). This is very frustrating and I have done this with both of my PS3 controller, always plugged in (haven't even messed with bluetooth yet).

Any help would be greatly appreciated! Thank you very much

gizmo98, 2016-04-08:
It's a sdl2.0.4 problem. We use sdl2.0.3 now again. Please update your setup script and run "sudo ./retropie_packages.sh sdl2".

closed 2016-04-08 by gizmo98

rsrobinson87, 2016-04-08:
ahhh, thank you so much!

rsrobinson87, 2016-04-08:
Ok, i updated the setup script, and when i run that command line, it tells me that the command is not found. what should i do now?

gizmo98, 2016-04-08:
Take a look here:
http://blog.petrockblock.com/forums/topic/problems-getting-ps3-controller-to-work-properly/#post-119984

rsrobinson87, 2016-04-08:
Thank you very much, my friend! It is all compiling/installing now. Crossing my fingers!