RetroPie/RetroPie-Setup #1421: Generic PS3 controller >

peer1979, 2016-04-08:
I've bought a generic ps 3 controller that will not connect with Retropie. It is detected by Windows 7 and Ubuntu 15.04 without issues, but fails underneath retropie.

This is a screenshot that shows the issue, it fails with 'can't set operational mode: step 3':
[image]

I've come accros somehting similar her: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1326725

Is it fixable or do you need some extra information?

joolswills, 2016-04-09:
What PSU are you using ? DId you google it - I googled it and found

https://www.raspberrypi.org/forums/viewtopic.php?t=34058 - so try it with a powered hub. It is best to use the forum first for controller problems as often it is connected to things like power etc.

peer1979, 2016-04-09:
Hello Jools,

Thanks for replying. I've tried the forum two times, but no answers :(

Because i've seen this message related to a bug, i figured it might be one. I've got an original raspbarry power supply with a raspberry pi 3. I don't own a powered usb hub, but am willing to buy one if it's needed. How can i make sure this is the problem?

Here are some additional screens, perhaps they help to figure this out:

lsusb:

[image]

hciconfig:

[image]

sixpair output:
[image]

joolswills, 2016-04-09:
I didn't see your posts on the old forum - I assume you didn't post to the new forum though as I've not seen it.

I don't know if the powered hub will help,and I can't be sure. I don't know which PSU that is, please provide details.

peer1979, 2016-04-11:
No worries, glad somebody is reacting now :) I did not repost on the new forum.

PSU: https://www.raspberrypi.org/products/universal-power-supply/
Power supply: I'm asking some friends if they have one, maybe i'm lucky.

But do you think this is connected to the bug report i've mentioned?

Dark-Show, 2016-04-12:
You said generic PS3 controller. Is it Sony brand or a Chinese knock-off? If a knock-off is there any information as to company or such. Maybe even a picture of the controller itself. Also are you using it VIA USB or Bluetooth?

peer1979, 2016-04-13:
I don't know if it is cheap, but it is a knock-off for sure.

Controller: Envivo PS3 controller, model 1262
Raspberry pi: It is recognized as 'Shengic 2009/07082XW-V1Inc.'

I've tried to follow procedure: Connect controller with usb and then disconnect and try to pair with bluetooth. It does not work both ways. The screenshot in the start topic is when it was connected with usb.

My controller:
[image]
[image]

Dark-Show, 2016-04-13:
This controller is not supported by sixad, If it ran with a gasia chip like a lot of clones do, you could enable hacks and get it to work. However, I'm not sure if the gasia hacks would help you with this controller.

peer1979, 2016-04-15:
Thanks for the response, now i finally know :) And you are right, if i'm one of few with this controller it is not worth the time spent.

I will try your suggestions and see what will happen.

closed 2016-06-10 by joolswills