RetroPie/RetroPie-Setup #1423: Generate gamepad configuration for Kodi 16 from pipplware repo

joolswills, 2016-04-09:
the Kodi 16 in pipplware repo has a patch for working gamepad support - we can now automatically generate a Kodi configuration based on the ES config, so people can navigate with their joypad.

HerbFargus, 2016-04-09:
I've been thinking about this a bit and I'm not entirely sure there is a clean way of doing this, the amount of controller variables is kind of ridiculous: see http://kodi.wiki/view/Keymap. to confuse it further, what it seemed like from my testing is the jstest values didnt necessarily reflect the values in the keymaps xml file (at least with the xbox userspace driver) a quick hacky method I used was to set up a repo with a few of the most common controller xmls and then just added a quick bit at the end of the module to install the xmls from my repo. see https://github.com/zerojay/RetroPie-Extra/pull/98  

I don't know if it would be easier to just add some basic navigation with xmls of the most common controllers unless there is some coding wizardry that can sort it dynamically like the other scripts.

diogomsantos, 2016-04-09:
Some joysticks are plug'n'play. Xbox 360 should be, take a look at:https://github.com/xbmc/xbmc/tree/Jarvis/system/keymaps. At least I can confirm that logitech, ps3, ps4, xbox 360 and some Saitek are. Some of this keymaps, support multiple comandos, see de xbox 360 for example.
Kodi 17 (will) support even more. See here:
https://github.com/kodi-game/peripheral.joystick/tree/master/peripheral.joystick/resources/buttonmaps/xml/linux

joolswills, 2016-04-09:
@HerbFargus there are a lot of options, but a basic config should be possible with the most used controls.

joolswills, 2016-04-09:
if many are supported out the box, perhaps we can not bother anyway :)

HerbFargus, 2016-04-09:
Their out of the box support applies to xpad, not to xboxdrv. And when installed there aren't any keymaps added, none of my controls worked out of the box.  I can look into maybe adding some stuff upstream so there is out of the box support for xboxdrv at least

diogomsantos, 2016-04-09:
The keymaps are installed in /usr/share/kodi/system/keymaps/ and are automatically loaded if the devices name matches any of <name> or <altname> arguments inside the .xml files.
Yeah, the xboxdrv seems to use different device names:
http://kodi.wiki/view/Xbox_360_Wireless_Controller

DuckSalmon, 2016-10-27:
With my gamesir controller linked as XBOX 360 controller i cannot use kodi
the menu remain on image option, like in this video
https://www.youtube.com/watch?v=ZCyrcVJtYCw

any solution? (without disable controllers)

joolswills, 2016-10-28:
@DuckSalmon please use the forum for support - https://retropie.org.uk/forum/

joolswills, 2017-04-04:
No longer needed as Kodi 17 has joypad configuration built in.

closed 2017-04-04 by joolswills