RetroPie/RetroPie-Setup #1426: Raspberry Pi - Pointer stuck at top left corner

zettam, 2016-04-12:
I've followed the readme file instructions in the official raspberry pi download (1.8.0). Made all of the steps, including the sld section.

However when I run scummvm (./scummvm) all I see is a fullscreen scummvm, with the mouse stuck at top left corner.
alt-enter also does not do anything. Not sure if this is a complete crash, or, only mpouse pointer not working issue.

HerbFargus, 2016-04-13:
The scummvm with retropie works fine, if you are building it manually ask for help on the forum as this is not a bug as requested in the opening text when you opened the issue.

https://retropie.org.uk/forum/

closed 2016-04-13 by HerbFargus

joolswills, 2016-04-13:
To note: retropie includes custom builds of sdl1/2 with changes, and it is possible that the problems are due to the way you are building it or the libraries you are building against - try our image or install via retropie-setup, but feel free to discuss on our forum https://retropie.org.uk/forum/