RetroPie/RetroPie-Setup #1428: Linux Headers and GameCon installation fails

semaf, 2016-04-13:
Linux Headers 4.1.19 and Gamecon installation fails while it gives a 404 error during the download of the Linux-Headers with the RetroPi-Setup.

The URL here is correct: https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/linux-headers-4.1.19-v7%2B_4.1.19-v7%2B-2_armhf.deb

joolswills, 2016-04-13:
Just tested here and it downloaded it ok - please make sure you have the latest version of retropie-setup and post a log and steps to reproduce if you can still see the problem.

semaf, 2016-04-13:
I am using berryboot-20160313-pi2-pi3.zip and downloading RetroPi directly from there. After all updates and RetroPi Update over RetroPi-Setup is the Error still there. Will try uninstalling and installing it over Terminal again.

joolswills, 2016-04-13:
BerryBoot have their own kernel version which is differently named - you will have to do it manually in that case as the automatic install is for Raspbian.

semaf, 2016-04-13:
Automated one does not work too.. I have installed the headers from the url above and still not working with the command "modprobe: FATAL: Module gamecon_gpio_rpi not found"

joolswills, 2016-04-13:
on Raspbian ?

joolswills, 2016-04-13:
I don't think this will work on berryboot TBH. use the Raspbian image - we don't support the BerryBoot image I'm afraid - it is provided due to demand only, but as BerryBoot have a kernel shared between the various distros, some stuff will not work out of the box.

semaf, 2016-04-13:
So installing Raspian and afterward RetroPi is better.. Jessie is not a Problem right?

HerbFargus, 2016-04-13:
retropie is built on raspbian. just use the image provided

joolswills, 2016-04-13:
You can do it that way, or use our premade image which is based on Jessie.

semaf, 2016-04-13:
I will give a try.. Hope it will work :)

One more question which has not to do with the topic: I could find Pinouts for NES Controllers but not that for GPIOs. Which GPIO are needed to be connected that RetroPi will detect the NES Controller?

joolswills, 2016-04-13:
Please use the forum for further help. Thanks.

joolswills, 2016-04-13:
I will add a warning dialog to the gamecondriver with some information if it can't find the kernel headers.

closed 2016-04-13 by joolswills

semaf, 2016-04-13:
Used without a reply!
https://retropie.org.uk/forum/topic/95/nes-controller-to-raspberry-pi-2-gpio

joolswills, 2016-04-13:
Maybe no-one has the answer yet - I have no idea regarding nes pinouts - I would have to google it.

joolswills, 2016-04-13:
I googled "nes gpio raspberry pi " and found https://www.raspberrypi.org/forums/viewtopic.php?f=78&t=79703 as the first match - if that is of any use.