RetroPie/RetroPie-Setup #1431: Rice Plugin Black Screen

HerbFargus, 2016-04-14:
This is more or less the same issue listed here: 
https://github.com/mupen64plus/mupen64plus-user-issues/issues/663

if you change the audio plugin in `/opt/retropie/configs/n64/mupen64plus.cfg`

```
# Filename of audio plugin
AudioPlugin = "mupen64plus-audio-sdl.so"
```
to
```
# Filename of audio plugin
AudioPlugin = "mupen64plus-audio-omx.so"
```

it will work. perhaps we should automatically switch to omx when the rice plugin is chosen as a workaround til the issue is fixed if its an upstream issue.

Also the `mupen64plus.cfg` is set as user root which is likely due to the chown being prior to this line:

https://github.com/RetroPie/RetroPie-Setup/blob/590fd796e5f253067ce96a31ac057f3761c1f816/scriptmodules/emulators/mupen64plus.sh#L134

gizmo98, 2016-04-14:
In my opinion this is already fixed upstream. mupen64plus package has to be rebuild.

joolswills, 2016-04-14:
I will rebuild the binaries and move the chown.

joolswills, 2016-04-14:
I rebuilt the binaries.

HerbFargus, 2016-04-14:
I can confirm this is fixed, thanks.

closed 2016-04-14 by HerbFargus