RetroPie/RetroPie-Setup #1435: Texas Instruments TI-99 Emulator Support

stevetb, 2016-04-15:
I would like to put in a request a Texas Instruments TI-99 Emulator to be added into RetroPie.  I found a link to several emulators which may work or already be supported.  ##

http://www.99er.net/emul.shtml

Please let me know if I can do anything to assist in moving this into an experimental build.

Thank you,
Stevetb

HerbFargus, 2016-04-15:
Already been done on zerojays repo:

https://github.com/zerojay/RetroPie-Extra/blob/master/scriptmodules/emulators/ti99sim.sh

stevetb, 2016-04-18:
Why is this not added to the main RetroPie Experimental Install Section?  This was a huge system back in the 1980's, seems worthy to me.

HerbFargus, 2016-04-18:
We are working on some fundamental changes to the menu before we add anymore new systems, but you can easily install the extra modules from zerojay's repository in the meantime: https://github.com/zerojay/RetroPie-Extra

stevetb, 2016-04-18:
Thank you HerbFargus.  I went through the installer and had some errors tonight.  I will go through the logs and see if it is related to 3.7 and submit to ZeroJay.  I appreciate the help.  Please give the developers a nudge to add if you can :-)

HerbFargus, 2016-04-18:
I am one of the developers ;)

stevetb, 2016-04-18:
Be gentle on yourself then, hahaha

HerbFargus, 2016-06-08:
added as of https://github.com/RetroPie/RetroPie-Setup/pull/1514

closed 2016-06-08 by HerbFargus