RetroPie/RetroPie-Setup #1444: Noobs install

procount, 2016-04-20:
I'm trying to create a NOOBS install of retropie to help someone out.
After noobs installs it and the first boot script runs, it seems to hang on resizing the partition. But this resizing is not necessary on a noobs install so I want to remove it.
Can you point me to this initial boot script please?

joolswills, 2016-04-20:
When opening a ticket it clearly states to use the forum for support

https://retropie.org.uk/forum/

That said - this is actually included in the documentation anyway. Googling "noobs retropie" brings back https://github.com/retropie/retropie-setup/wiki/convert-retropie-sd-card-image-to-noobs-image which contains info about this.

closed 2016-04-20 by joolswills

joolswills, 2016-04-20:
We get post after post on the issue tracker with support requests, which is frustrating since we added the contributing guide, then we added a template which needs removing before posting, and yet some people still ignore this.

The forum is the perfect place for help, as it has many users who use it daily. Even when there are things thought of as "bugs", often it is linked with a particular hardware combination or configuration, which is far more likely to be fixed if you can find someone else with the same setup. This is why we ask users to post to the forum first for help. With >400,000 downloads of RetroPie-Setup v3.6 we would not be able to manage support requests on this issue tracker.

procount, 2016-04-20:
Sorry. My google karma failed me. Thanks for quick response