RetroPie/RetroPie-Setup #1449: Random freezing when using bluetooth ps3 controllers on pi 3 Retropie 3.7

tommydills, 2016-04-23:
I'm having some frequent freezing issues with emulationstation, only when I'm using bluetooth controllers.  Sometimes it is in game, sometimes when exiting a game, and sometimes when starting a game.  I am using the onboard bluetooth, and unfortunately I do not have a dongle to try instead of the onboard.  Last night it probably happened about 5 times in 2 hours, and I am usually unable to ssh into the machine, but if i do it wont follow through with the shutdown command.

joolswills, 2016-04-23:
Known hardware/driver issue - https://github.com/raspberrypi/linux/issues/1360 - there are multiple threads about this on the forum also.

joolswills, 2016-04-23:
The solution currently is to use an additional bluetooth dongle.

tommydills, 2016-04-23:
whoops, i looked for it but must have missed it.  Thanks.

joolswills, 2016-04-23:
It's ok to leave this open for now - will hopefully aid with people looking for the problem. cheers.

dcs619, 2016-04-23:
+1 same issue

oguzhan1907, 2016-04-23:
+1 same same

joolswills, 2016-04-24:
I'm not sure there is any benefit doing a +1 - it is a known issue that affects all PI3 users. Adding +1 will not help it be fixed any quicker.

HerbFargus, 2016-04-24:
Also to mention there are emoticon +1 things that you can put on the original post without spamming the thread

rajder, 2016-04-25:
Sorry for the spam, just showing my interest in this issue to get notifications on updates. :)

eddywebs, 2016-05-05:
+1 issue with pi3 crashing with onboard bluetooth and ps3 controllers, is there any status updated on this ?

joolswills, 2016-05-05:
Follow the earlier link (which would be clearer if everyone wasn't adding +1 posts which don't help)

dankcushions, 2016-05-11:
looks like this might be fixed in kernel 4.4 https://github.com/raspberrypi/linux/issues/1360#issuecomment-217863895 (and later updates)

konradsa, 2016-05-13:
It's pretty much confirmed now that this bug has been fixed in kernel 4.4. To the Retropie devs: Please update Retropie to a newer kernel version so that everybody can get the benefit. Based on reports from [https://github.com/raspberrypi/linux/issues/1360](https://github.com/raspberrypi/linux/issues/1360), kernel 4.4 works without a problem with Retropie 3.7. This post also has instructions on how to upgrade  the kernel manually for everybody here that wants to try out the fix.

joolswills, 2016-05-13:
RetroPie is not a not a distro - our image is just RetroPie install on top of Raspbian. So when Raspbian update the packaged kernel, our next image will contain it.

konradsa, 2016-05-13:
Looks like this just happened actually:
https://www.raspberrypi.org/blog/another-update-raspbian/

eddywebs, 2016-05-15:
There are no more crashes after updating to linux kernel to 4.4 >> https://github.com/raspberrypi/linux/issues/1360#issuecomment-217863895

However any additional controller still fail to pair on retropie, this time there is no bluetooth crash its just no paring as 2nd or 3d controller.

RetroPieNerd, 2016-05-26:
You can close this issue now...

I have updated the Bluetooth page: https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-Bluetooth-controller

closed 2016-06-10 by joolswills