RetroPie/RetroPie-Setup #1451: Add player-statistics , level-up-system/ranks/badges/achievements

Micr0Bit, 2016-04-23:
player could earn experience and unlock ranks or badges/achievements by pressing a specific controller-button "x" times ... playing x-amount of time on a system/game ... and so on 
could have that level-up-system visually appear as a "local profile-page"

zerojay, 2016-04-24:
https://retro-acheivements.org

Micr0Bit, 2016-04-24:
@zerojay : thx , i had exactly this in my mind when thinking about this feature ... but could not remember the website anymore :) (this and steam) 
* http://retroachievements.org

and here an example for a profile with badges : 
http://retroachievements.org/User/KanaMojo

HerbFargus, 2016-04-24:
We won't be implementing anything like this outside of the functionality retroarch already provides.

Micr0Bit, 2016-04-24:
retroach looks more of a dead end tbh. -> http://retroachievements.org/viewtopic.php?t=9

well i would love to see something similar for retropie , but i understand that its gonna happen from "here" ... maybe someone picks this up and gets interested

closed 2016-04-24 by Micr0Bit

Micr0Bit, 2016-04-24:
oh my bad sry ... i see "retroarch" now ... im gonna check that out , thx guys

HerbFargus, 2016-04-24:
For the record it's in the documentation:

https://github.com/retropie/retropie-setup/wiki/RetroAchievements