RetroPie/RetroPie-Setup #1454: 3.7 Samba Shares not working in OSX

craigerskine, 2016-04-26:
Did the Samba protocols get updated in 3.7? It looks like there is an incompatibility with OSX as my shares are not connecting anymore. Others are seeing this too: https://discussions.apple.com/thread/7534687

joolswills, 2016-04-26:
It's possible there was a Raspbian security update that has affected it. Note that RetroPie sits on top of Raspbian, so it is not something that is directly linked to RetroPie development.

you should `sudo apt-get update; sudo apt-get dist-upgrade` to see if any further updates help - but if it is a compatibility issue, there is nothing to do about it our end - perhaps there are some samba options that you could change though. I have no idea, and do not run OS X.

JCrew7384, 2016-06-14:
I tried updating and unable to connect via shared drive in Mac or via smb. SSH through Mac works though.

HerbFargus, 2016-06-14:
Tried updating what? Supposedly if you update samba to 4.4.3 it resolves it (https://discussions.apple.com/message/30176289#30176289) there are also workarounds referenced on the aforementioned thread. I don't have a Mac so I can't test it.

JCrew7384, 2016-06-14:
Please ignore my noob status, how do I update Samba on the Raspberry Pi?

joolswills, 2016-06-14:
you can't. you can work around the samba issue by adding a password for the pi user, and using that user for the shares. You might do best to ask on that apple forum if you need help to do this.

JCrew7384, 2016-06-14:
I'll have to re read it. Maybe I mis understood. Isn't the password for the pi user just raspberry? I will check when back at my computer. Thanks for the comments.

fmsarubbi, 2016-06-23:
The fix mentioned in the forum fixed it for me. 
https://retropie.org.uk/forum/topic/569/samba-apt-get-upgrade-samba-is-not-working-anymore/4

joolswills, 2016-07-11:
Is this still an issue with the latest samba on Raspbian ? This post suggests it has been sorted https://www.raspberrypi.org/forums/viewtopic.php?p=998743#p998743

joolswills, 2016-07-11:
If it is, I will enable the "pi" account for samba in the next image with the default password (or you can do it manually yourself with `echo -ne "raspberry\nraspberry\n" | sudo smbpasswd -s -a pi`

prosthetichead, 2016-07-17:
issue was in my install i did last night adding the defualt pi user as jollswills said fixed issue and i can connect using smb now using pi/raspberry

joolswills, 2016-07-20:
@prosthetichead please can you give me more details - did you update all the raspbian packages also ?

prosthetichead, 2016-07-21:
Sorry.

No I didn't update any of the raspbian packages. It was a fresh install of retroPie, I tried using SMB shares to access the Pi but it would not accept a guest or the default pi/raspberry login. Quick google and I found this issue and your suggestion of running  `echo -ne "raspberry\nraspberry\n" | sudo smbpasswd -s -a pi` which enabled the pi/raspberry login and SMB has worked without issue since then on my mac.

joolswills, 2016-07-22:
It would be useful if someone can test this with the latest raspbian samba packages without the workaround to confirm if it is still an issue or not.

dankcushions, 2016-07-22:
Can confirm it's working in OSX without the workaround (which I never did). I have:
samba Version: 2:4.2.10+dfsg-0+deb8u3
samba-common-bin Version: 2:4.2.10+dfsg-0+deb8u3

not sure when i last updated but apparently I have the latest versions of both of these

closed 2016-07-23 by joolswills

dustinbolton, 2016-08-15:
Still not working with the latest RetroPie image on the site for Raspberry Pi 3 ( https://github.com/RetroPie/RetroPie-Setup/releases/download/3.8.1/retropie-v3.8.1-rpi2_rpi3.img.gz ).

dankcushions, 2016-08-15:
The fix isn't in that image, which was released before it.

diamondsw, 2016-12-12:
Fix still isn't in a fresh download of v4.1. Fail.

joolswills, 2016-12-12:
@diamondsw it has been confirmed fixed by other users. If you need help please use the RetroPie forum https://retropie.org.uk/forum/

diamondsw, 2016-12-15:
Apologies; just set up a brand new system with the 4.1 image from the site, fully updated via apt-get update/upgrade - and it didn't work until I found my way here and added the password via smbpasswd. I'll refer to the forums in the future.

olivierlefloch, 2018-11-10:
I believe this is still an issue (latest OS X, latest retropie with all packages up to date).