RetroPie/RetroPie-Setup #1457: Python script run from within RetroPie-Setup does not show text output

gizmo98, 2016-05-01:
I just tried to add an extra configuration module for reicast. There is a input configuration script named reicast-joyconfig. If i run this script from within RetroPie-Setup there is no text output like "You have x controllers, select...". Strange enough installation logs show this output. Do i miss something?
'''
rp_module_id="reicastinput"
rp_module_desc="Configure input devices for Reicast"
rp_module_menus="3+configure"
rp_module_flags="nobin"

function configure_reicastinput() {
    file="/tmp/temp.cfg"
    /opt/retropie/emulators/reicast/bin/
    reicast-joyconfig -f /tmp/temp.cfg
    iniConfig " = " "" "$file"
    iniGet "mapping_name"
    mv "$file" "$configdir/dreamcast/mappings/controller_${ini_value// /}.cfg"
}
'''

joolswills, 2016-05-07:
Did you try running it manually via `retropie_packages.sh` rather than via retropie_setup (which does some redirecting which could break the output due to buffering)

joolswills, 2016-05-07:
(and if so, you can try adding `>/dev/tty` to the command which may help)

gizmo98, 2016-05-07:
Thanks Jools! reicast_packages.sh runs fine. With `>/dev/tty`the script can be used with retropie_setup.sh.

```
rp_module_id="reicastinput"
rp_module_desc="Configure input devices for Reicast"
rp_module_menus="3+configure"
rp_module_flags="nobin"

function configure_reicastinput() {
    file="/tmp/temp.cfg"
    /opt/retropie/emulators/reicast/bin/
    clear
    reicast-joyconfig -f /tmp/temp.cfg > /dev/tty
    iniConfig " = " "" "$file"
    iniGet "mapping_name"
    mv "$file" "$configdir/dreamcast/mappings/controller_${ini_value// /}.cfg"
}
```

closed 2016-05-07 by gizmo98