RetroPie/RetroPie-Setup #1458: Bluetooth configuration and conflict joypad-keyboard

KZL92, 2016-05-01:
Hello I noticed a trouble with the bluetooth package. Seems to do not be related to RetroPie, but still a problem.

1) First, the daemon is not started by default.. I don't know why
Someone can confirm? It's a fresh setup from my side.

```pi@retropie:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
Active: inactive (dead)
Docs: man:bluetoothd(8)

pi@retropie:~ $ sudo /etc/init.d/bluetooth start
[ ok ] Starting bluetooth (via systemctl): bluetooth.service.

pi@retropie:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
Active: active (running) since Tue 2016-04-26 23:42:07 UTC; 1s ago
Docs: man:bluetoothd(8)
Main PID: 1028 (bluetoothd)
Status: "Running"
CGroup: /system.slice/bluetooth.service
└─1028 /usr/lib/bluetooth/bluetoothd```

The solution that I found is to type this command, to force starting the daemon at some point :
echo "/etc/init.d/bluetooth start" >> /etc/rc.local

2) Major problem is the configuration of retroarch, using the GUI on EmulationStation>RetroPie. The retroarch application is not working properly. I connected one keyboard & one 8bitdo controller at the same time. It's really a mess to make the both working at the same time because when you configure the 8bitdo controller, it's modifying the keyboard mapping. (cf. [https://retropie.org.uk/forum/topic/511/8bitdo-not-detected-on-emulationstation/14](https://retropie.org.uk/forum/topic/511/8bitdo-not-detected-on-emulationstation/14) [https://retropie.org.uk/forum/topic/511/8bitdo-not-detected-on-emulationstation/14](https://retropie.org.uk/forum/topic/511/8bitdo-not-detected-on-emulationstation/13)). Btw I noticed that you cannot change the number max of user, it's going back to the default value or I don't know.. Seems completely bugged.

3) The wiki here is not updated, some commands are not necessary to make the bluetooth adapters working. Btw, concerning the 8bitdo files for example there is some default files available you just need to put this file (https://github.com/paalfe/mixedcontent/blob/master/udev_rules.d/99-8bitdo-bluetooth-controllers.rules) here to make EmulationStation worked : /etc/udev/rules.d/ for example.

I hope it's helping..
Cheers

joolswills, 2016-05-09:
bluetooth is started by default - must be something wrong with your install. There would be many more bug reports if it wasn't on by default on our image (you don't say if you used the image, or installed on top of existing system)

not sure I understand the problem in issue 2 - but it sounds like a configuration issue

3) I don't have one of these controllers - others who do have contributed to the wiki and it does have information on how to pair. It doesn't include example lines for every controller no - but feel free to edit the wiki and reference that file.

joolswills, 2016-05-09:
BTW if you have previously paired a PS3 controller from retropie-setup you will need to remove/disable that as currently you can't have that and the built in Bluetooth stack running at the same time.

closed 2016-05-16 by joolswills