RetroPie/RetroPie-Setup #1462: New linapple issues. (easy fixes..)

dabonetn, 2016-05-06:
Ok, I haven't tried a new PI only install, but trying it on the pc I see 2 problems.

The Linapple fails on pc compile. It's missing libsdl-image1.2-dev
This is a added dependency not in the original linapple.

The launch options is emulators.cfg should be changed to 
linapple="/opt/retropie/emulators/linapple/linapple -r -f -1 %ROM%"

The -r is autorun the emulation, without this it sits and waits for the run key (F2) to be pressed.
The -f is run full screen (The pi is always full screen, the pc needs this so it doesn't just open a window.)

Thanks.
Later,
dabone

HerbFargus, 2016-05-06:
@dabonetn  would enabling joystick 1 by default cause any issues because currently the script enables both joysticks on install. no sense in scripting it if you have it set upstream

dabonetn, 2016-05-06:
I only have one enabled, because most games only supported one joystick. 
There was only one joystick port on the machine.

Just let me know what changes you want to my repo.

Later,
dabone

HerbFargus, 2016-05-06:
well if it doesnt hurt to have both enabled may as well, it will simplify the module.

As an aside when I was testing your changes it still required a keyboard to start up my pacman game and choose a joystick but I suppose that is to be expected as it was originally a computer. all the same its an improvement from how it used to be.

dabonetn, 2016-05-06:
That depends on the program, some games just require a joystick. Choplifter is an example.

dabonetn, 2016-05-06:
But because my Pi is built into a computer, I always have a keyboard handy..

[http://www.lemon64.com/forum/viewtopic.php?t=56679](url)

HerbFargus, 2016-05-06:
I see. 

I've made some changes feel free to test.

https://github.com/RetroPie/RetroPie-Setup/pull/1463

and feel free to enable the second joystick by default (not sure if it makes a difference or not.)

HerbFargus, 2016-05-06:
Added as of https://github.com/RetroPie/RetroPie-Setup/pull/1463

closed 2016-05-06 by HerbFargus