RetroPie/RetroPie-Setup #1466: Vice errors on x11 install (PC x86)

dabonetn, 2016-05-07:
ok, I've been trying retropie on xubuntu and see the following vice errors.

It's prefixing /lib to the install directory so it's installing vice to /opt/retropie/emulators/vice/lib/
(Under the amd64 it's the same)

But the emulators.cfg is pointing to:

vice-x64="/opt/retropie/emulators/vice/bin/x64 %ROM%"

Also the sdl-vicerc that is created only contains the line
[C64]

joolswills, 2016-05-07:
I made an error when last updating this module to work on 64bit Ubuntu/X11 - I have duplicated the prefix line, but I suspect I meant to rename one of the options. Will sort thanks.

joolswills, 2016-05-07:
Should be ok now - must have been a c&p fail or some test code - removed the additional prefix line. on 64bit lib is symlinked to lib64, as otherwise it doesn't find the system roms etc.

commit 196d10b6 (2016-05-07): vice changes - fixes #1466
 * remove duplicated configure --prefix (c&p fail / debug or something)
 * disable ffmpeg support, as it can fail building against recent ffmpeg (at least on Ubuntu 16.04)
closed 2016-05-07 by joolswills