RetroPie/RetroPie-Setup #1477: Cheats don't work in Picodrive or Genesis-Plus-GX

deusxm, 2016-05-11:
Cheat databases load absolutely fine to give a menu option, but once enabled and applied via the Retroarch GUI menu, do not actually activate in-game.

This issue does not happen with cores for other systems.

zerojay, 2016-05-14:
Make sure you don't have retro achievements set to hardcore mode as it disables the ability to cheat and use save states.

deusxm, 2016-05-14:
I can confirm I'm not using RetroAchievements. Save states work fine.

joolswills, 2016-05-20:
https://github.com/libretro/picodrive/issues/25

amigagamer, 2016-09-25:
Genesis Plus GX retroarch core does appear to work with cheats, however you need the game genie codes on the few iive tried (retroarch cheats files are the pro action replay codes)
Heres an example thats worked for me- Battle Squadron.  Create (or edit the existing) Battle Squadron.cht with the following.
`cheats = 2 

cheat0_desc = "Infinite Ships P1"
cheat0_code = "AD1T-AAE0"
cheat0_enable = true 

cheat1_desc = "Master Code"
cheat1_code = "R1JT-R60T"
cheat1_enable = true `

joolswills, 2017-05-31:
Also fixed upstream with picodrive now.

closed 2017-05-31 by joolswills